leggedrobotics/darknet_ros #1: Undefined Reference to cvRound when compiling with OpenCV 3

zhenkaiwang, 2017-04-26:
I am trying to build darknet_ros under Ubuntu16.06 with OpenCV 3.2 and I met this error::

[ 49%] Building CXX object darknet_ros/darknet_ros/CMakeFiles/darknet_ros.dir/src/object_detector_demo.cpp.o
[ 49%] Building C object darknet_ros/darknet_ros/CMakeFiles/darknet_ros.dir/src/image_interface.c.o
[ 50%] Linking CXX executable /media/sdcard/catkin_ws/devel/lib/darknet_ros/darknet_ros
CMakeFiles/darknet_ros.dir/__/darknet/src/convolutional_layer.c.o: In function `cvPointFrom32f':
convolutional_layer.c:(.text+0x660): undefined reference to `cvRound'
convolutional_layer.c:(.text+0x670): undefined reference to `cvRound'
CMakeFiles/darknet_ros.dir/__/darknet/src/convolutional_layer.c.o: In function `cvReadInt':
convolutional_layer.c:(.text+0x1214): undefined reference to `cvRound'
CMakeFiles/darknet_ros.dir/__/darknet/src/convolutional_layer.c.o: In function `cvEllipseBox':
convolutional_layer.c:(.text+0x14b8): undefined reference to `cvRound'
convolutional_layer.c:(.text+0x14d0): undefined reference to `cvRound'
CMakeFiles/darknet_ros.dir/__/darknet/src/deconvolutional_layer.c.o:deconvolutional_layer.c:(.text+0x660): more undefined references to `cvRound' follow
CMakeFiles/darknet_ros.dir/__/darknet/src/deconvolutional_layer.c.o: In function `make_deconvolutional_layer':
deconvolutional_layer.c:(.text+0x1b18): undefined reference to `forward_deconvolutional_layer_gpu'
deconvolutional_layer.c:(.text+0x1b1c): undefined reference to `forward_deconvolutional_layer_gpu'
deconvolutional_layer.c:(.text+0x1b24): undefined reference to `backward_deconvolutional_layer_gpu'
deconvolutional_layer.c:(.text+0x1b28): undefined reference to `backward_deconvolutional_layer_gpu'
deconvolutional_layer.c:(.text+0x1b30): undefined reference to `update_deconvolutional_layer_gpu'
deconvolutional_layer.c:(.text+0x1b34): undefined reference to `update_deconvolutional_layer_gpu'
CMakeFiles/darknet_ros.dir/__/darknet/src/lsd.c.o: In function `cvPointFrom32f':
lsd.c:(.text+0x660): undefined reference to `cvRound'
lsd.c:(.text+0x670): undefined reference to `cvRound'
CMakeFiles/darknet_ros.dir/__/darknet/src/lsd.c.o: In function `cvReadInt':
lsd.c:(.text+0x1214): undefined reference to `cvRound'
CMakeFiles/darknet_ros.dir/__/darknet/src/lsd.c.o: In function `cvEllipseBox':
lsd.c:(.text+0x14b8): undefined reference to `cvRound'
lsd.c:(.text+0x14d0): undefined reference to `cvRound'
CMakeFiles/darknet_ros.dir/__/darknet/src/image.c.o:image.c:(.text+0x660): more undefined references to `cvRound' follow
collect2: error: ld returned 1 exit status
darknet_ros/darknet_ros/CMakeFiles/darknet_ros.dir/build.make:3384: recipe for target '/media/sdcard/catkin_ws/devel/lib/darknet_ros/darknet_ros' failed
make[2]: *** [/media/sdcard/catkin_ws/devel/lib/darknet_ros/darknet_ros] Error 1
CMakeFiles/Makefile2:2796: recipe for target 'darknet_ros/darknet_ros/CMakeFiles/darknet_ros.dir/all' failed
make[1]: *** [darknet_ros/darknet_ros/CMakeFiles/darknet_ros.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2

Any idea how to solve it?

mbjelonic, 2017-04-27:
The code should be compatible with OpenCV 3.2.

are you trying to compile darknet_ros with or without cuda (using GPU)? If you are using your graphics card, please try it again because I added a missing file to the CMakeLists.txt.

zhenkaiwang, 2017-04-27:
Yes, I am compiling with cuda 8.0 on Jetson TX2. I have tried it again using new CMakeLists.txt. But still got the same error...

> [ 98%] Building C object darknet_ros/darknet_ros/CMakeFiles/darknet_ros.dir/__/darknet/src/regressor.c.o
In file included from /opt/ros/kinetic/include/opencv-3.2.0-dev/opencv2/core/core_c.h:48:0,
                 from /opt/ros/kinetic/include/opencv-3.2.0-dev/opencv2/highgui/highgui_c.h:45,
                 from /media/sdcard/catkin_ws/src/darknet_ros/darknet/src/image.h:13,
                 from /media/sdcard/catkin_ws/src/darknet_ros/darknet/src/network.h:5,
                 from /media/sdcard/catkin_ws/src/darknet_ros/darknet/src/detector.c:1:
/opt/ros/kinetic/include/opencv-3.2.0-dev/opencv2/core/types_c.h: In function ‘cvPointFrom32f’:
/opt/ros/kinetic/include/opencv-3.2.0-dev/opencv2/core/types_c.h:929:13: warning: implicit declaration of function ‘cvRound’ [-Wimplicit-function-declaration]
     ipt.x = cvRound(point.x);
             ^
In file included from /opt/ros/kinetic/include/opencv-3.2.0-dev/opencv2/core/core_c.h:48:0,
                 from /opt/ros/kinetic/include/opencv-3.2.0-dev/opencv2/highgui/highgui_c.h:45,
                 from /media/sdcard/catkin_ws/src/darknet_ros/darknet/src/image.h:13,
                 from /media/sdcard/catkin_ws/src/darknet_ros/darknet/src/network.h:5,
                 from /media/sdcard/catkin_ws/src/darknet_ros/darknet/src/regressor.c:1:
/opt/ros/kinetic/include/opencv-3.2.0-dev/opencv2/core/types_c.h: In function ‘cvPointFrom32f’:
/opt/ros/kinetic/include/opencv-3.2.0-dev/opencv2/core/types_c.h:929:13: warning: implicit declaration of function ‘cvRound’ [-Wimplicit-function-declaration]
     ipt.x = cvRound(point.x);
             ^
[100%] Linking CXX executable /media/sdcard/catkin_ws/devel/lib/darknet_ros/darknet_ros
CMakeFiles/darknet_ros.dir/src/image_interface.c.o: In function `cvPointFrom32f':
image_interface.c:(.text+0x660): undefined reference to `cvRound'
image_interface.c:(.text+0x670): undefined reference to `cvRound'
CMakeFiles/darknet_ros.dir/src/image_interface.c.o: In function `cvReadInt':
image_interface.c:(.text+0x1214): undefined reference to `cvRound'
CMakeFiles/darknet_ros.dir/src/image_interface.c.o: In function `cvEllipseBox':
image_interface.c:(.text+0x14b8): undefined reference to `cvRound'
image_interface.c:(.text+0x14d0): undefined reference to `cvRound'
CMakeFiles/darknet_ros.dir/__/darknet/src/convolutional_layer.c.o:convolutional_layer.c:(.text+0x660): more undefined references to `cvRound' follow
collect2: error: ld returned 1 exit status
darknet_ros/darknet_ros/CMakeFiles/darknet_ros.dir/build.make:1957: recipe for target '/media/sdcard/catkin_ws/devel/lib/darknet_ros/darknet_ros' failed
make[2]: *** [/media/sdcard/catkin_ws/devel/lib/darknet_ros/darknet_ros] Error 1
CMakeFiles/Makefile2:2796: recipe for target 'darknet_ros/darknet_ros/CMakeFiles/darknet_ros.dir/all' failed
make[1]: *** [darknet_ros/darknet_ros/CMakeFiles/darknet_ros.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2

mbjelonic, 2017-04-30:
Hi,

maybe the following links can help you: 
https://github.com/opencv/opencv/issues/8438
https://groups.google.com/forum/#!topic/darknet/fQ2GQuibBA4

The error occurs in the official darknet code. Please have a look in the darknet google group to see how to fix this.

Best,
Marko

mbjelonic, 2017-05-01:
Add this to CMakeLists.txt:

add_definitions(-O4 -g)

closed 2017-05-01 by mbjelonic

ac-optimus, 2018-12-30:
@mbjelonic  "add_definitions(-O4 -g)" made my day. Thanks.