leggedrobotics/darknet_ros #2: undefined reference to `load_network_demo'

pirobot, 2017-04-27:
Hello,

I am trying to run catkin_make on a machine running Ubuntu 16.04, ROS Kinetic and CUDA-8.  I get the following error:

```
[ 51%] Linking CXX executable /home/patrick/catkin_ws/devel/lib/darknet_ros/darknet_ros
CMakeFiles/darknet_ros.dir/src/YoloObjectDetector.cpp.o: In function `darknet_ros::YoloObjectDetector::init()':
YoloObjectDetector.cpp:(.text+0x11fd): undefined reference to `load_network_demo'
collect2: error: ld returned 1 exit status
darknet_ros/darknet_ros/CMakeFiles/darknet_ros.dir/build.make:1856: recipe for target '/home/patrick/catkin_ws/devel/lib/darknet_ros/darknet_ros' failed
make[2]: *** [/home/patrick/catkin_ws/devel/lib/darknet_ros/darknet_ros] Error 1
CMakeFiles/Makefile2:3666: recipe for target 'darknet_ros/darknet_ros/CMakeFiles/darknet_ros.dir/all' failed
make[1]: *** [darknet_ros/darknet_ros/CMakeFiles/darknet_ros.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j1 -l1" failed

```

mbjelonic, 2017-04-30:
Hi,
I do not know why this error occurs. I will try to build the repo with CUDA-8 in the next week and see if I get the same issue. 
-Marko

pirobot, 2017-05-02:
OK, thanks.  It builds fine on Ubuntu 14.04 and ROS Indigo so we are using that for now.

mbjelonic, 2017-05-03:
Ok, thanks for the update. I will look into this issue as soon as possible. I will let you know if I have more news.

malcolmwhat, 2017-06-06:
Did you ever look into this?

mbjelonic, 2017-06-07:
Yes, but I do not know where this is coming from.

malcolmwhat, 2017-06-07:
I got it to build without seeing this issue. I got a cuda error because my version of gcc and g++ were too recent so I switched to gcc-5 and g++-5 and now both `catkin build` and `catkin_make -j1 -l1` work.

I don't actually have a GPU on my laptop, so the tests don't pass unless I remove cuda from my system and rebuild.

closed 2020-11-27 by tomlankhorst