leggedrobotics/darknet_ros #6: Compile error on 16.04 ubuntu

tdnet12434, 2017-05-01:
I clone the latest darknet with working well alone with GPU=1 and CV=1.
Then i download this repository and linked the darknet path that is on my home directory.
Below is reported error. There are also have some warning cvround but i skip it all for clear vision.

```
/home/fx/catkin_ws/src/darknet_ros/darknet_ros/src/object_detector_demo.cpp: In function ‘void* detect_in_thread(void*)’:
/home/fx/catkin_ws/src/darknet_ros/darknet_ros/src/object_detector_demo.cpp:99:48: error: invalid conversion from ‘int’ to ‘float**’ [-fpermissive]
     get_region_boxes(l, in.w, in.h, net.w, net.h, demo_thresh, probs, boxes, 0, 0, demo_hier, 1);
                                                ^
/home/fx/catkin_ws/src/darknet_ros/darknet_ros/src/object_detector_demo.cpp:99:96: error: cannot convert ‘float’ to ‘box*’ for argument ‘6’ to ‘void get_region_boxes(layer, int, int, float, float**, box*, int, int*, float, int)’
     get_region_boxes(l, in.w, in.h, net.w, net.h, demo_thresh, probs, boxes, 0, 0, demo_hier, 1);
```

It would be appriciate for any suggestion. 
Thank you advance.

mbjelonic, 2017-05-01:
Hi,

The cvround warnings are coming from the original darknet code. 

Nevertheless, I think I know the issue in your error. You are using an oder version of darknet somehow. Make sure that you are cloning my repo with the right submodule! Clone my repo by following this:

```
cd catkin_workspace/src
git clone --recursive git@github.com:leggedrobotics/darknet_ros.git
cd ../
catkin build darknet_ros
```

In older commits of darknet the function get_region_boxes was implemented like this: 

`void get_region_boxes(layer l, int w, int h, float thresh, float **probs, box *boxes, int only_objectness, int *map, float tree_thresh);`

Nowadays, the function looks like this: 

`void get_region_boxes(layer l, int w, int h, int netw, int neth, float thresh, float **probs, box *boxes, int only_objectness, int *map, float tree_thresh, int relative);`

Make sure that you are cloning the right submodule because I am using the new version of get_region_boxes.

-Marko

tdnet12434, 2017-05-01:
Thank you for quick reply ! 

I tried that you suggestion. It bring me to opencv error below.
```
CMakeFiles/darknet_ros.dir/src/image_interface.c.o: In function `cvPointFrom32f':
image_interface.c:(.text+0x54d): undefined reference to `cvRound'
image_interface.c:(.text+0x563): undefined reference to `cvRound'
CMakeFiles/darknet_ros.dir/src/image_interface.c.o: In function `cvReadInt':
image_interface.c:(.text+0xe68): undefined reference to `cvRound'
CMakeFiles/darknet_ros.dir/src/image_interface.c.o: In function `cvEllipseBox':
image_interface.c:(.text+0x10d4): undefined reference to `cvRound'
image_interface.c:(.text+0x10f6): undefined reference to `cvRound'
CMakeFiles/darknet_ros.dir/home/fx/catkin_ws/src/darknet_ros/darknet/src/convolutional_layer.c.o:convolutional_layer.c:(.text+0x54d): more undefined references to `cvRound' follow
collect2: error: ld returned 1 exit status
make[2]: *** [/home/fx/catkin_ws/devel/.private/darknet_ros/lib/darknet_ros/darknet_ros] Error 1
make[1]: *** [CMakeFiles/darknet_ros.dir/all] Error 2
make: *** [all] Error 2

```

i notice that is the same https://github.com/leggedrobotics/darknet_ros/issues/1 issue.
I try to make with some modified flag as https://groups.google.com/forum/#!topic/darknet/fQ2GQuibBA4
it used to fix the DEBUG=1 flag issue but not for darknet_ros. It still produce the same error.

tdnet12434, 2017-05-01:
I success to build it now by add the same definition as darknet make file in CMakelist.txt

`add_definitions(-O4 -g)`

closed 2017-05-01 by tdnet12434

mbjelonic, 2017-05-01:
Nice, thanks! I will add this to the CMakelist.txt.