leggedrobotics/darknet_ros #7: Memory leak issue

tdnet12434, 2017-05-01:
Hi @mbjelonic 
I encounter the issue about memory that continue rising without bound. 
But when i using darknet alone it not happen.
How to avoid this? and any idea would be appreciate.
Thank you advance.

mbjelonic, 2017-05-03:
Ok, thanks for the update. I will look into this issue as soon as possible. I will let you know if I have more news.

mbjelonic, 2017-05-10:
@tdnet12434 : I have created a repo called fix/memory_leak. I already tested it on two different machines and it seems that memory usage is not increasing. Can you please check on your computer and give me feedback? If it works,  I will merge it to master. Thanks!

Best,
Marko

tdnet12434, 2017-05-10:
Thank you for update :D 
I tried with my both labtop front camera and image from kinect. But it fail when it bring up first image frame with this error.

```
[ INFO] [1494431817.460666931]: [YoloObjectDetector] USB image received.
[ INFO] [1494431817.539748487]: [YoloObjectDetector] runYolo().
CUDA Error: invalid device function
darknet_ros: /home/fx/catkin_ws/src/darknet_ros-fix-memory_leak/darknet/src/cuda.c:36: check_error: Assertion `0' failed.

```

tdnet12434, 2017-05-10:
Sorry for above error report which is from my own mistake about CUDA flag.
(forgot to change -gencode arch )

It's work perfectly now!!
Thank you again :+1: 

Best,
Thanabadee

closed 2017-05-10 by tdnet12434

mbjelonic, 2017-05-10:
Thank you for the update! I am glad that it is working!

Best,
marko

Abduoit, 2017-07-26:
@tdnet12434 
Hi all
I am using ubuntu 16.04, Nvidia Quadro K600, CUDA 8, NVIDIA-SMI 375.66
I do not know how to change the ARCH in Makefile, would you please write down for me, I tried this but it failed

`ARCH= --gpu-architecture=compute_30, --gpu-code=compute_30`

when I run

./darknet detector demo cfg/coco.data cfg/yolo.cfg yolo.weights

I got this error

```
21 conv    512  1 x 1 / 1    19 x  19 x1024   ->    19 x  19 x 512
   22 CUDA Error: out of memory
darknet: ./src/cuda.c:36: check_error: Assertion `0' failed.
Aborted (core dumped)
```

mbjelonic, 2017-07-27:
@Abduoit: You are not using the ROS version of this repo. The darknet folder is only a submodule that points to the original repo of darknet. You could use also my repo to start darknet but I recommend you to use the code https://github.com/pjreddie/darknet and if you still have an issue please post it in the original repo. If you would like to start my repo in ROS, then you need to follow my readme.

pushkalkatara, 2018-10-18:
@tdnet12434 How did you resolve the issue of memory rising out of bounds. I also tried changing the architecture to sm_53 on Nvidia Jetson TX1, but still facing the same issue.