leggedrobotics/darknet_ros #8: cameraCallback not getting called

halt9, 2017-05-06:
Hi,

I have a node publishing at `chameleon3/image_raw/compressed` but despite setting my launch file properly, I am not getting a callback. I have modified my launch file as such:
```
<arg name="camera_topic_name" default="/chameleon3/image_raw"/>
<arg name="image_transport" default="compressed"/>
```

But all I get when I run the node (via `roslaunch src/darknet_ros/darknet_ros/launch/darknet_ros.launch` since I don't have catkin installed) I get a black screen, and the `ROS_INFO` message for the callback method being called never prints. Do I have a configuration issue?

mbjelonic, 2017-05-06:
Hi, 
the subscriber expects a sensor_msgs::Image but probably your topic is a sensor_msgs/CompressedImage. You first need to republish your topic into the right format. You uncomment the last line in my launch file and change it to:

`<node name="republish" type="republish" pkg="image_transport" output="screen" 	args="compressed in:=/chameleon3/image_raw/compressed raw out:=/chameleon3/image_raw" />`

Then it should work if you use:

`<arg name="camera_topic_name" default="/chameleon3/image_raw"/>`

You do not need the line: 

<arg name="image_transport" default="compressed"/>

Best,
Marko

closed 2017-05-06 by mbjelonic

halt9, 2017-05-06:
You're right! It works, however, with a minor adjustment:
`<node name="republish" type="republish" pkg="image_transport" output="screen" args="compressed in:=/chameleon3/image_raw/ raw out:=/chameleon3/image_decompressed" />`