leggedrobotics/darknet_ros #10: Cannot use other weight, cfg files.

yukkysaito, 2017-06-05:
This ros wrapper is great.

This wrapper cannot load other weight, cfg files?
I generate a weight and cfg files using other training data (not VOC).
In original darknet, yolo detect a object using the files.

But this node cannot detect using the files.

mbjelonic, 2017-06-07:
Yes, that is true. I planned to integrate the possibility to load other weights and cfg files. You need to change some of my code in order to make this work or you just wait till I implement this. :)

mbjelonic, 2017-06-28:
I am currently adding this new functionality inside the branch devel/change_class_labels. The implementation is already finished and I verified the changes with the getests. But I still need to test this properly with a camera and different weight and cfg files. Nevertheless, you can already test this. You need to provide the config and weight file inside darknet_ros/cfg and darknet_ros/weight. In addition, you need to create a yaml file inside darknet_ros/config and use this new yaml file in your launch file.

mbjelonic, 2017-06-29:
It seems to work. I also added the COCO detection objects. Have fun!

closed 2017-06-29 by mbjelonic