leggedrobotics/darknet_ros #12: darknet path not found

getaleks, 2017-06-25:
Hi,

I'm getting an error that my the darknet path cannot be found:

```
CMake Error at /usr/share/cmake-3.5/Modules/FindCUDA.cmake:1693 (add_executable):
  Cannot find source file:

    DARKNET_PATH-NOTFOUND/src/gemm.c

  Tried extensions .c .C .c++ .cc .cpp .cxx .m .M .mm .h .hh .h++ .hm .hpp
  .hxx .in .txx
Call Stack (most recent call first):
  darknet_ros/darknet_ros/CMakeLists.txt:83 (cuda_add_executable)

-- Generating done
-- Build files have been written to: /home/user/catkin_ws/build
Invoking "cmake" failed
```

Is there a fix for this? Or can I supply the darknet path myself somehow?

Thanks in advance!

mbjelonic, 2017-06-25:
Hi,

read the readme. You need to clone this repo recursively. 

Best,
Marko

getaleks, 2017-06-25:
Hi Marko,

thanks for the quick reply. I was unable to clone recursively because of a key error, but its fixed now.

Unfortunately, I still seem to be missing some files. BoundingBoxes.h is missing in darknet_ros_msgs?

Here is the whole error message again:

`In file included from /home/user/catkin_ws/src/darknet_ros/darknet_ros/src/YoloObjectDetector.cpp:10:0:
/home/user/catkin_ws/src/darknet_ros/darknet_ros/include/darknet_ros/YoloObjectDetector.h:34:44: fatal error: darknet_ros_msgs/BoundingBoxes.h: No such file or directory
compilation terminated.
darknet_ros/darknet_ros/CMakeFiles/darknet_ros.dir/build.make:139: recipe for target 'darknet_ros/darknet_ros/CMakeFiles/darknet_ros.dir/src/YoloObjectDetector.cpp.o' failed
make[3]: *** [darknet_ros/darknet_ros/CMakeFiles/darknet_ros.dir/src/YoloObjectDetector.cpp.o] Error 1
make[3]: *** Waiting for unfinished jobs....
In file included from /home/user/catkin_ws/src/darknet_ros/darknet_ros/src/object_detector_demo.cpp:15:0:
/home/user/catkin_ws/src/darknet_ros/darknet_ros/include/darknet_ros/YoloObjectDetector.h:34:44: fatal error: darknet_ros_msgs/BoundingBoxes.h: No such file or directory
compilation terminated.
darknet_ros/darknet_ros/CMakeFiles/darknet_ros.dir/build.make:187: recipe for target 'darknet_ros/darknet_ros/CMakeFiles/darknet_ros.dir/src/object_detector_demo.cpp.o' failed
make[3]: *** [darknet_ros/darknet_ros/CMakeFiles/darknet_ros.dir/src/object_detector_demo.cpp.o] Error 1
In file included from /home/user/catkin_ws/src/darknet_ros/darknet_ros/src/yolo_object_detector_node.cpp:10:0:
/home/user/catkin_ws/src/darknet_ros/darknet_ros/include/darknet_ros/YoloObjectDetector.h:34:44: fatal error: darknet_ros_msgs/BoundingBoxes.h: No such file or directory
compilation terminated.
darknet_ros/darknet_ros/CMakeFiles/darknet_ros.dir/build.make:163: recipe for target 'darknet_ros/darknet_ros/CMakeFiles/darknet_ros.dir/src/yolo_object_detector_node.cpp.o' failed
make[3]: *** [darknet_ros/darknet_ros/CMakeFiles/darknet_ros.dir/src/yolo_object_detector_node.cpp.o] Error 1
CMakeFiles/Makefile2:2859: recipe for target 'darknet_ros/darknet_ros/CMakeFiles/darknet_ros.dir/all' failed
make[2]: *** [darknet_ros/darknet_ros/CMakeFiles/darknet_ros.dir/all] Error 2
CMakeFiles/Makefile2:2871: recipe for target 'darknet_ros/darknet_ros/CMakeFiles/darknet_ros.dir/rule' failed
make[1]: *** [darknet_ros/darknet_ros/CMakeFiles/darknet_ros.dir/rule] Error 2
Makefile:1200: recipe for target 'darknet_ros' failed
make: *** [darknet_ros] Error 2
Invoking "make darknet_ros -j4 -l4" failed
`

akshayshetty, 2017-07-12:
I was able to build it successfully. I followed these steps:

- Clone recursively:
`git clone --recursive https://github.com/leggedrobotics/darknet_ros.git`

- In the CMakeLists for darknet_ros, remove `-std=c++11` from line 22. Also change your compute capability accordingly on line 23.

- Build the package:
`catkin_make -DCMAKE_BUILD_TYPE=Release`

aarontan94, 2018-05-30:
@akshayshetty hello, I did as you described however I am now getting an error that reads:

CMake Error at darknet_ros_lib_generated_maxpool_layer_kernels.cu.o.cmake:207 (message):
  Error generating
  /home/aaron/catkin_ws/build/darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/./darknet_ros_lib_generated_maxpool_layer_kernels.cu.o

darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/build.make:126: recipe for target 'darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/darknet_ros_lib_generated_maxpool_layer_kernels.cu.o' failed
make[2]: *** [darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/darknet_ros_lib_generated_maxpool_layer_kernels.cu.o] Error 1
CMakeFiles/Makefile2:5905: recipe for target 'darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/all' failed
make[1]: *** [darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/all] Error 2

Do you know what this means?

JaimeGJ, 2018-06-23:
Hi @aarontan94 
Could you resolve that problem with the maxpool_layer_kernels? I'm having the same error... :(

aarontan94, 2018-06-23:
@JaimeGJ I have no idea what that means and I have yet to figure this out ...

JiyongOh, 2018-10-24:
I avoid the same error by cloning darknet sources into the folder /catkin_workspace/darknet_ros/darknet.

frannecki, 2019-03-03:
hi @aarontan94 @JaimeGJ  so do you ever found out the solution to that "Error generating" problem? I'm having the same issue and I'm using ros kinect on ubuntu 16.04

frannecki, 2019-03-21:
I skipped downloading the weights by directly copying into the dirs and somehow it worked

closed 2020-11-27 by tomlankhorst