leggedrobotics/darknet_ros #20: How to get image coordinates for located objects?

ElisabethE, 2017-07-19:
Hi! I'm wondering how I get the image coordinates of the classified objects (e.g. that box drawn around the object found).

See the attached log for the information available to me, 

If I can get those image coordinates, I could find the distance to said object using a Kinect, and move further from there. It's the last piece I need.

I'm thankful for any ideas you might have because the documentation says nothing about this... :(

[darknet_ros_output.txt](https://github.com/leggedrobotics/darknet_ros/files/1159440/darknet_ros_output.txt)

mbjelonic, 2017-07-20:
See the chapter "Published Topics" in the readme. I am publishing an array of bounding boxes in image coordinates. You just need to subscribe in your code to the topic and then you have everything that you need.

closed 2017-07-20 by mbjelonic

catalin15dcm, 2018-08-13:
@ElisabethE how do you get the coordinates? 
I'm trying to get those coordinates but am a beginner in ros and i don't put it to an end. Could you help me?