leggedrobotics/darknet_ros #21: Floating point exception(core dumped) when run dartknet_ros

mzwang828, 2017-08-09:
Hey,
I got the package compiled but when I run the darknet_ros node, I got this:

```
[ INFO] [1502249531.341824814]: [YoloObjectDetector] Node started.
[ INFO] [1502249531.343953332]: [YoloObjectDetector] Xserver is running.
[ INFO] [1502249531.344425343]: [YoloObjectDetector] init().
Floating point exception (core dumped)
```
I have CUDA 8, ubuntu 14.04 and indigo.

Do you have any clue how to solve this problem? Thank!

mbjelonic, 2017-08-09:
Hi, 

what exactly are you running? This error is coming from the function YoloObjectDetector::init() of darknet_ros/src/YoloObjectDetector.cpp Can you insert some couts in this function and check which line is causing the error?

This problem can occur, if you have no detection objects in your detection config file, i.e. the array of the ROS parameter yolo_model/detection_classes/names is empty.

Best,
Marko

mzwang828, 2017-08-09:
Thanks for the quick reply!
I don't know what I did wrong last night, but it works very well now. Thanks for your help and the package!

closed 2017-08-09 by mzwang828