leggedrobotics/darknet_ros #23: Cannot catkin_make

Kn15263ss, 2017-08-10:
Hi~ I have a little problem
I use ubuntu 14.04 , ROS:indigo
When i catkin_make, i encounter a problem:
[image]
[image]

How to fix this problem?
Thanks you!

mbjelonic, 2017-08-11:
Hi, 
I think the gnu compiler 4.8 does not fully support c++11. Please try to use gnu 4.9.2 or higher. For reference: I am using version 5.4.0 on my tx2.
Best,
Marko

Kn15263ss, 2017-08-13:
I have install gcc version 4.8.5 and g++ version 7.1.0, and I don't know what is gnu.

mbjelonic, 2017-08-14:
Check out this: https://gcc.gnu.org/

Updating gcc is exactly what I meant. Is it still not working?

Kn15263ss, 2017-08-14:
Yes! It is still not working

Hothaifa, 2017-09-11:
I had the same problem and i'm still using gcc 4.8. you can fix it by adding --std=c++11 to NVCC_FLAG in the CMakeLists.txt. i created a pull request about this problem

BolinZhou, 2017-10-23:
I think this has to do with the darknet architecture supports. I think the original darknet had commoned out the line `-gencode arch=compute_20,code=[sm_20,sm_21]` of the [ darknet_ros/darknet_ros/CMakeLists.txt](url) try it without that line

closed 2020-11-27 by tomlankhorst