leggedrobotics/darknet_ros #25: FPS issue

joyyang1215, 2017-08-11:
Hi,
I've tried original darknet on TX1 for tiny-yolo model for 12FPS
but this source only 6FPS
Is there any optimization option that I miss?
thanks

mbjelonic, 2017-08-11:
Hi, 
can you please send me the launch file and the config file that you are using? How do you calculate the 6 FPS?
Best,
Marko

joyyang1215, 2017-08-14:
Hi,
here's my person_darknet_ros.launch
`<?xml version="1.0" encoding="utf-8"?>

<launch>
  
  <arg name="yolo_weights_path"          default="$(find darknet_ros)/yolo_network_config/weights"/>
  <arg name="yolo_config_path"           default="$(find darknet_ros)/yolo_network_config/cfg"/>

  
  <rosparam command="load" ns="darknet_ros" file="$(find darknet_ros)/config/ros.yaml"/>
  <rosparam command="load" ns="darknet_ros" file="$(find darknet_ros)/config/person_tiny_yolo_voc.yaml"/>

  
  <node pkg="darknet_ros" type="darknet_ros" name="darknet_ros" output="screen">
    <param name="weights_path"          value="$(arg yolo_weights_path)" />
    <param name="config_path"           value="$(arg yolo_config_path)" />
  </node>

  
  <node name="usb_cam" pkg="usb_cam" type="usb_cam_node" output="screen" >
    <param name="video_device" value="/dev/video0" />
    <param name="image_width" value="640" />
    <param name="image_height" value="480" />
    <param name="pixel_format" value="yuyv" />
    <param name="camera_frame_id" value="usb_cam" />
    <param name="io_method" value="mmap"/>
  </node>
</launch>
`
And the person_tiny_yolo_voc.yaml
`image_view:

  enable_opencv: true
  use_darknet: false
  wait_key_delay: 2

yolo_model:

  config_file:
    name: person_tiny-yolo-voc.cfg
  weight_file:
    name: person_tiny-yolo-voc_40000.weights
  threshold:
    value: 0.5
  detection_classes:
    names:
      - person
`
Best,
Joy

gftabor, 2017-08-31:
I am getting similar performance as he described using yolo. Everything is running out of the box with usb webcam.

  <arg name="yolo_weights_path"          default="$(find darknet_ros)/yolo_network_config/weights"/>
  <arg name="yolo_config_path"           default="$(find darknet_ros)/yolo_network_config/cfg"/>

  
  <rosparam command="load" ns="darknet_ros" file="$(find darknet_ros)/config/ros.yaml"/>
  <rosparam command="load" ns="darknet_ros" file="$(find darknet_ros)/config/yolo_voc.yaml"/>

  
  <node pkg="darknet_ros" type="darknet_ros" name="darknet_ros" output="screen">
    <remap from="/darknet_ros/camera_reading" to="/usb_cam/image_raw"/>

    <param name="weights_path"          value="$(arg yolo_weights_path)" />
    <param name="config_path"           value="$(arg yolo_config_path)" />
  </node>

YoshiRi, 2017-09-28:
Hi. I have same issue with my jetson TX1 Ubuntu 16.04.
My frame rate is about 6 FPS and I tried some solution but not satisfying.

My version of cuda is 8.0 and cudnn is version 5, and video rate is 30FPS with 640*480.
I am now using tiny-yolo-voc with config width and height is both 288.
Also, I tried catkin_make with Release mode.

Here is a last part of my ros.yaml file.
`
image_view:
  enable_opencv: false
  use_darknet: true
  wait_key_delay: 1
  enable_console_output: false
`
And are there any other way to get faster performance?

Thank you,
Yoshi

joyyang1215, 2017-10-17:
I've tried with 
```
enable_opencv: false 
use_darknet: true
```
on Jetson TX2 for 9.8FPS.
And with 
```
enable_opencv: true
use_darknet: false
```
get 8.2FPS for average.
But, original darknet c code can run on 14FPS with tiny-yolo model.

mbjelonic, 2017-11-20:
Hi,

can you please checkout the new branch called "devel/threads" and try to run it again? I heavily rearranged the code and this should also improve the speed of the detection algorithm by adding new threads. Please note that the branch is still under development and I did not add all ROS publishers in this branch. I just wanted to see if this setup improves the speed of the detection. You can check the update rate in the terminal.

Please let me know if the new code design improves also the speed in your case. Thanks in advance!

Best,
Marko

DanLiu0623, 2018-01-15:
@joyyang1215 Hi, I met the same issue with you, Did you solve it?

DanLiu0623, 2018-01-15:
@mbjelonic  It works with branch "devel/threads",but  it can not show detect image,I think it is a little problem,Thank you very much~

mbjelonic, 2018-01-15:
@DanLiu0623 : I know about this and I still need to finish the branch. I am more interested about the frame rate in this branch. Adding the ros publishers is easy in the end.

DanLiu0623, 2018-01-15:
@mbjelonic It can reach 20fps with tiny-yolo of darknet_ros, and only 8fps with master, as same as darknet detects.

mbjelonic, 2018-01-15:
@DanLiu0623 : I am more interested in the update rate of devel/threads. Thanks.

DanLiu0623, 2018-01-15:
@mbjelonic Yes, 20fps is the result of branch "devel/threads",and what I git clone is the newest branch "devel/threads".

mbjelonic, 2018-01-15:
@DanLiu0623 : Ok perfect! So it is basically the same like the darknet repo. I will finish the implementation on the branch devel/threads in 2 weeks and then I will merge into master. After that you will have the same functionalities like the one in master.

DanLiu0623, 2018-01-16:
@mbjelonic Hi, could you give me some information how to fix the issue that why it could't show detect image normally?

mbjelonic, 2018-01-16:
I had to restructure the code a lot in order to achieve the frame rate. I have to add the ros publishers again and also make sure that this is thread safe. On the branch devel/threads the reading and detection of the images are on two different threads now. I just need some time to cleanup everything. 

You can see the images in the opencv image view if you plugin a monitor to the tx2 and run the code without ssh to the tx2. That is how I do it with my tx2 to see the detection images. As soon as I integrated the ros publishers again, this will not be needed anymore.

DanLiu0623, 2018-01-17:
@mbjelonic I see your code in detail, you comment the run_yolo function, so it means that you did't open detect,so how could I get fps? maybe what I got is not true detect fps, right?

mbjelonic, 2018-01-29:
Merged the branch devel/threads.

closed 2018-01-29 by mbjelonic

gftabor, 2018-12-01:
@sharoseali what is the resolution of the images? 

Also maybe worth including results from "nvidia-smi" and "top".

sharoseali, 2018-12-03:
@sharoseali what is the resolution of the images?
> 
> Also maybe worth including results from "nvidia-smi" and "top".
@gftabor 
The input is video in mp4 format with 1080p resolution. 
what is this ? 
Also maybe worth including results from "nvidia-smi" and "top"

gftabor, 2018-12-03:
So your 3fps seems pretty reasonable. The other data from the thread was 20 fps using 480p.

https://www.google.com/search?client=ms-unknown&ei=Rn0FXJ_6Horh0gK2wK-AAw&q=1080+*+1920+%2F%28640*480%29&oq=1080+*+1920+%2F%28640*480%29&gs_l=mobile-gws-wiz-serp.3..19.2171.10950..12343...0.0..0.205.1343.2j8j1......0....1.........0i71j0i13i30j33i160.lSZgPcPEZA4

And if we assume the execution time is linear to total pixels 21/7=3