leggedrobotics/darknet_ros #26: Error generating cuda files?

mattmakesthings, 2017-08-11:
Hi I'm running into an issue when building using  

catkin build darknet_ros -DCMAKE_BUILD_TYPE=Release

darknet_ros_msgs builds just fine but when it gets to darknet_ros I get this error
[image]

mbjelonic, 2017-08-11:
Hi,

Can you please tell me the GPU that you are using? It might be a problem with the compute capability (version). This means that you need to check the compute capability (version) of your GPU. You can find a list of supported GPUs in CUDA here: https://en.wikipedia.org/wiki/CUDA#Supported_GPUs. Simply find the compute capability of your GPU and replace in darknet_ros/CMakeLists.txt the version number 61 with your version number:

-gencode arch=compute_61,code=sm_61

Best,
Marko

mattmakesthings, 2017-08-11:
Hi Marko thanks for responding!
Im using a gtx 980 and have already made that change after looking over the past posts
[image]

mbjelonic, 2017-08-11:
Please undo your changes in the CMakeLists and try again. I already added the code generation for the compute capability 5.2. Can you please send me mor information if it still not working? E.g. Ubuntu version, ROS version and GNU version.

mattmakesthings, 2017-08-12:
Still having same issue. I'm using: 
Ubuntu 16.04
ROS Kinetic
Linux version 4.10.0-30-generic
gcc version 5.4.0
Cuda compilation tools, release 8.0, V8.0.61

mattmakesthings, 2017-08-21:
@mbjelonic would you mind posting what your versions of cuda, cmake, and gcc you were using?

mbjelonic, 2017-08-22:
Hi, I tested darknet_ros on the following systems:

Ubuntu 14.04 and 16.04
ROS Kinetic and Indigo
Linux Version 4.10.0-27-generic
gcc version 5.4.0
Cuda compilation tools, release 8.0, V8.0.61
Graphics cards: Titan X (Compute capability 6.1) and Jetson Tx2 (Compute capability 6.2)
cmake version 3.5.1

Your system is similar to mine. You can try the following:

Build darknet_ros with CUDA and run it on your CPU. Simply copy and paste the following into your CMakeLists.txt:

`cmake_minimum_required(VERSION 2.8.12)
project(darknet_ros)

# Set c++11 cmake flags
set(CMAKE_CXX_FLAGS "-std=c++11 ${CMAKE_CXX_FLAGS}")

# Define path of darknet folder here.
find_path(DARKNET_PATH
  NAMES "README.md"
  HINTS "${CMAKE_CURRENT_SOURCE_DIR}/../darknet/")
message("Darknet path dir = ${DARKNET_PATH}")
add_definitions(-DDARKNET_FILE_PATH="${DARKNET_PATH}")
list(APPEND LIBRARIES "m")

# Find X11
MESSAGE ( STATUS "Searching for X11..." )
FIND_PACKAGE ( X11 REQUIRED )
IF ( X11_FOUND )
  INCLUDE_DIRECTORIES ( ${X11_INCLUDE_DIR} )
  LINK_LIBRARIES ( ${X11_LIBRARIES} )
  MESSAGE ( STATUS " X11_INCLUDE_DIR: " ${X11_INCLUDE_DIR} )
  MESSAGE ( STATUS " X11_LIBRARIES: " ${X11_LIBRARIES} )
ENDIF ( X11_FOUND )

# Find rquired packeges
find_package(Boost REQUIRED COMPONENTS thread)
find_package(OpenCV REQUIRED)
find_package(catkin REQUIRED
  COMPONENTS
    cv_bridge
    roscpp
    rospy
    std_msgs
    actionlib
    darknet_ros_msgs
    image_transport
)

# Enable OPENCV in darknet
add_definitions(-DOPENCV)
add_definitions(-O4 -g)

catkin_package(
  INCLUDE_DIRS
    include
  LIBRARIES
    ${PROJECT_NAME}_lib
  CATKIN_DEPENDS
    cv_bridge
    roscpp
    actionlib
    rospy
    std_msgs
    darknet_ros_msgs
    image_transport
  DEPENDS
[... 87 lines cut ...]
## Testing ##
#############

if(CATKIN_ENABLE_TESTING)
  # Download yolo-voc.weights
  set(PATH "${CMAKE_CURRENT_SOURCE_DIR}/yolo_network_config/weights")
  set(FILE "${PATH}/yolo-voc.weights")
  message(STATUS "Checking and downloading yolo-voc.weights if needed ...")
  if (NOT EXISTS "${FILE}")
    message(STATUS "... file does not exist. Downloading now ...")
    execute_process(COMMAND wget http://pjreddie.com/media/files/yolo-voc.weights -P ${PATH})
  endif()

  find_package(rostest REQUIRED)

  # Object detection in images.
  add_rostest_gtest(${PROJECT_NAME}_object_detection-test
    test/object_detection.test
    test/test_main.cpp
    test/ObjectDetection.cpp
  )
  target_link_libraries(${PROJECT_NAME}_object_detection-test
    ${catkin_LIBRARIES}
  )
endif()`

mattmakesthings, 2017-09-10:
So it ended up actually being an error with my cuda installation

I followed these steps and got darknet_ros to run
https://covijn.com/2016/11/fixing-darknet-opencv3-make-error-convolutional_kernels/

mbjelonic, 2017-09-11:
Thanks for the update!

closed 2017-09-11 by mbjelonic

He-Rong, 2019-03-13:
I met the same problem and I am wondering more details to solve the problem. Did you reinstall cuda?

cardboardcode, 2019-08-14:
@mbjelonic thanks, your suggestion worked for me. 

```
Build darknet_ros without CUDA and run it on your CPU. 
Simply remove the following line in the CMakeLists.txt:

FIND_PACKAGE(CUDA)
```
To be more specific, please comment out the lines **16** and **17** of  `CMakeLists.txt` under `darknet_ros` ROS package.

line 16: [find package(CUDA QUIET)](https://github.com/leggedrobotics/darknet_ros/blob/ac666ab8e8e3dd23a8a95d891fb90874e63c8cb5/darknet_ros/CMakeLists.txt#L16)
line 18: [find package(CUDA REQUIRED)](https://github.com/leggedrobotics/darknet_ros/blob/ac666ab8e8e3dd23a8a95d891fb90874e63c8cb5/darknet_ros/CMakeLists.txt#L18)