leggedrobotics/darknet_ros #28: Error a launch

fuloporsi, 2017-08-17:
Hi! After a successful build, I have this error, when I try roslaunch:

core service [/rosout] found
ERROR: cannot launch node of type [darknet_ros/darknet_ros]: can't locate node [darknet_ros] in package [darknet_ros]
No processes to monitor
shutting down processing monitor...
... shutting down processing monitor complete

Ubuntu 14.04
ROS Indigo
I'm trying to use it without Cuda.
Any help would be appreciated.

pfankhauser, 2017-08-17:
Szia Orsi! =)
Did you use `catkin build` to compile? If yes, you should resource the workspace with `source ~/catkin_ws/devel/setup.bash` before running the node. You can find more about this in our [ROS course](http://www.rsl.ethz.ch/education-students/lectures/ros.html), specifically [these slides](https://www.ethz.ch/content/dam/ethz/special-interest/mavt/robotics-n-intelligent-systems/rsl-dam/ROS2017/lecture1.pdf) on slide nr. 22.
Let me know if this solves the issue.

fuloporsi, 2017-08-17:
Hey! :)
I tried resourcing it too, the issue still persists. And yes, I used `catkin build darknet_ros -DCMAKE_BUILD_TYPE=Release
` to compile.

pfankhauser, 2017-08-17:
Can you double check if it build darknet_ros correctly? It does not seem find it, so maybe it wasn't built correctly.

fuloporsi, 2017-08-17:
Actually it is giving me warnings. I can attach a screenshot from where it begins.
[image]

samuelba, 2017-08-17:
Do you build with install space enabled? Can you post your output for `catkin config`.

fuloporsi, 2017-08-18:
I enabled install space. Same error. 
Here is my `catkin config` output.
[image]

samuelba, 2017-08-18:
I just ask because the package is missing some install commands and doesn't run with install space enabled. But it should run without install space.

fuloporsi, 2017-08-21:
Hey! :)
I managed to start it from QT creator. I still don't know what was the problem, but at least it's working now. 
Thanks for the tips!

mbjelonic, 2017-08-22:
Great!

closed 2017-08-22 by mbjelonic