leggedrobotics/darknet_ros #30: The first image is treated differently than the following

Sagexs, 2017-09-04:
hi,
I trained yolo with a custom dataset. When I tried to use this package, only the first image after startup was correctly classified. This image was also displayed smaller. All following pictures were enlarged and no longer correctly classified.

mbjelonic, 2017-09-05:
Hi, 

can you please check which viewer you are using, i.e. image_view/use_darknet is true or false? Please try both options here. In addition, you can check if the darknet_ros topics are published. 

Please send me your yaml, .cfg and .weights file. 

Best,
Marko

Sagexs, 2017-09-05:
hi Marko,
thanks for the reply.

image_view/use_darknet is true or false --> same error for both
darknet_ros topics are published --> yes darknet_ros topics are published, first msg with found object all following topics with zero objects

Here the files: 
[weights and cfg](http://www.ilm.ing.tu-bs.de/blume/deepLearning/cfg_weights.zip)
[config.zip](https://github.com/leggedrobotics/darknet_ros/files/1276601/config.zip)
[testimage.jpg.zip](https://github.com/leggedrobotics/darknet_ros/files/1276621/testimage.jpg.zip)
[bagfile for test (ros indigo)](http://www.ilm.ing.tu-bs.de/blume/deepLearning/2017-08-31-13-51-41.bag)

Best,
Tobias

CristiIuga, 2017-09-14:
Hi Tobias

I have a similar bug with this darknet_ros package. I trained a network for 1 class and then tried to run it. Everything built correctly but I did not see any output at all. I tried with each viewer, the topics are published, but still no output.
If I test my network with the YOLO detection framework from the developers at [pjreddie.com/darknet/yolo/](https://pjreddie.com/darknet/yolo/), then it works.

Cristi

Sagexs, 2017-09-15:
Hi Cristi,
I couldn't find the error.
this ros yolo version works for me [link](https://github.com/ayushgaud/darknet_ros)

Best 
Tobias

closed 2021-05-19 by tomlankhorst