leggedrobotics/darknet_ros #31: No output for 1 class detection

CristiIuga, 2017-09-04:
Hi

I used this [blog post](https://timebutt.github.io/static/how-to-train-yolov2-to-detect-custom-objects/) to train a 1 class yolo network.
All is ok and I can run the network with _./darknet detector demo_ command and it works fine, the network outputs some bounding boxes.
When I make the changes to darknet_ros(created new yaml file, copied cfg and weights files to appropriate folders), the program compiles ok, runs ok, but I have no output. I tried changing the detection threshold to minimum and nothing, the number of detected bboxes is always 0. On the other hand, darknet_ros works fine with the standard yolo-voc.cfg and weights.
I attached the yaml and cfg files.
[cfg-yaml.zip](https://github.com/leggedrobotics/darknet_ros/files/1275083/cfg-yaml.zip)

mbjelonic, 2017-09-05:
Hi, 

can you please check which viewer you are using, i.e. image_view/use_darknet is true or false? Please try both options here. In addition, you can check if the darknet_ros topics are published.

Best,
Marko

CristiIuga, 2017-09-05:
Hi again

Thanks for the comment. Unfortunately it still doesn't output anything.
I was using the open_cv viewer but I just tried both now and is still the same. Using the yolo-voc, it works fine with both opencv and darknet viewers, I see detected bboxes. When I switch to my 1 class network, neither opencv or darknet viewers show anything, not even incorrect detections. I am looking at the /darknet_ros/found_object topic and the data is always 0.
Have you tried training a particular network and then test it with darknet_ros to see if it works? If so, could you give the link?

Cristi

CristiIuga, 2017-09-13:
Hello again

Did you try running darknet_ros with a specially trained network for 1 class for example? And did you have any results?
Also, besides yolo-voc and tiny yolo, have you tried training any other network and then use it in darknet_ros?

Best,
Cristi

mbjelonic, 2017-11-20:
Hi, 
can you pull the repo and try your network again? I changed the image encoding from BGR8 to RGB8. This might be the issue.
Best,
Marko

ChunJyeBehBeh, 2019-04-15:
@CristiIuga  Hello, do you manage to get the output at the end?

tomlankhorst, 2021-05-19:
Closing due to inactivity. Feel free to reopen.

closed 2021-05-19 by tomlankhorst