leggedrobotics/darknet_ros #33: Catkin build test error

Hothaifa, 2017-09-11:
when i try to run catkin build i get a error message saying this that bulid/devel file is used by catkin_make and asks to use different folder. I changed the folder to build2/devel2 but  i did not work 

```
catkin build darknet_ros --no-deps --verbose --catkin-make-args run_tests 
--------------------------------------------------------------------------
Profile:                     default
Extending:          [cached] /home/wavesha/catkin_ws/devel:/opt/ros/indigo
Workspace:                   /home/wavesha/catkin_ws 
--------------------------------------------------------------------------
Source Space:       [exists] /home/wavesha/catkin_ws/src
Log Space:          [exists] /home/wavesha/catkin_ws/logs
Build Space:        [exists] /home/wavesha/catkin_ws/./build2
Devel Space:        [exists] /home/wavesha/catkin_ws/./devel2
Install Space:      [unused] /home/wavesha/catkin_ws/install
DESTDIR:            [unused] None
--------------------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
--------------------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: run_tests
Internal Make Job Server:    True
Cache Job Environments:      False
--------------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
--------------------------------------------------------------------------
Workspace configuration appears valid.
--------------------------------------------------------------------------
[build] Found '2' packages in 0.0 seconds.                                     
[build] Package table is up to date.                                           
Starting  >>> darknet_ros                                                      
Starting   >> darknet_ros:loadenv                                              
Output     << darknet_ros:loadenv /home/wavesha/catkin_ws/logs/darknet_ros/build.loadenv.004.log
Loading environment from: /home/wavesha/catkin_ws/./devel2/env.sh
Finished   << darknet_ros:loadenv                                              
Starting   >> darknet_ros:mkdir                                                
Starting   >> darknet_ros:mkdir                                                
Starting   >> darknet_ros:cache-manifest                                       
Starting   >> darknet_ros:check                                                
Subprocess  > darknet_ros:check `cd /home/wavesha/catkin_ws/./build2/darknet_ros; catkin build --get-env darknet_ros | catkin env -si  /usr/bin/make cmake_check_build_system; cd -`
Output     << darknet_ros:check /home/wavesha/catkin_ws/logs/darknet_ros/build.check.003.log
/usr/bin/cmake -H/home/wavesha/catkin_ws/src/darknet_ros/darknet_ros -B/home/wavesha/catkin_ws/build2/darknet_ros --check-build-system CMakeFiles/Makefile.cmake 0
cd /home/wavesha/catkin_ws/./build2/darknet_ros; catkin build --get-env darknet_ros | catkin env -si  /usr/bin/make cmake_check_build_system; cd -
Finished   << darknet_ros:check                                                
Starting   >> darknet_ros:make                                                 
Subprocess  > darknet_ros:make `cd /home/wavesha/catkin_ws/./build2/darknet_ros; catkin build --get-env darknet_ros | catkin env -si  /usr/bin/make run_tests --jobserver-fds=6,7 -j; cd -`
Output     << darknet_ros:make /home/wavesha/catkin_ws/logs/darknet_ros/build.make.004.log
/usr/bin/cmake -H/home/wavesha/catkin_ws/src/darknet_ros/darknet_ros -B/home/wavesha/catkin_ws/build2/darknet_ros --check-build-system CMakeFiles/Makefile.cmake 0
/usr/bin/make -f CMakeFiles/Makefile2 run_tests
make[1]: Entering directory `/home/wavesha/catkin_ws/build2/darknet_ros'
/usr/bin/cmake -H/home/wavesha/catkin_ws/src/darknet_ros/darknet_ros -B/home/wavesha/catkin_ws/build2/darknet_ros --check-build-system CMakeFiles/Makefile.cmake 0
/usr/bin/cmake -E cmake_progress_start /home/wavesha/catkin_ws/build2/darknet_ros/CMakeFiles 3
/usr/bin/make -f CMakeFiles/Makefile2 CMakeFiles/run_tests.dir/all
make[2]: Entering directory `/home/wavesha/catkin_ws/build2/darknet_ros'
/usr/bin/make -f CMakeFiles/clean_test_results_darknet_ros.dir/build.make CMakeFiles/clean_test_results_darknet_ros.dir/depend
/usr/bin/make -f gtest/CMakeFiles/gtest.dir/build.make gtest/CMakeFiles/gtest.dir/depend
make[3]: Entering directory `/home/wavesha/catkin_ws/build2/darknet_ros'
cd /home/wavesha/catkin_ws/build2/darknet_ros && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/wavesha/catkin_ws/src/darknet_ros/darknet_ros /home/wavesha/catkin_ws/src/darknet_ros/darknet_ros /home/wavesha/catkin_ws/build2/darknet_ros /home/wavesha/catkin_ws/build2/darknet_ros /home/wavesha/catkin_ws/build2/darknet_ros/CMakeFiles/clean_test_results_darknet_ros.dir/DependInfo.cmake --color=
make[3]: Entering directory `/home/wavesha/catkin_ws/build2/darknet_ros'
cd /home/wavesha/catkin_ws/build2/darknet_ros && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/wavesha/catkin_ws/src/darknet_ros/darknet_ros /usr/src/gtest /home/wavesha/catkin_ws/build2/darknet_ros /home/wavesha/catkin_ws/build2/darknet_ros/gtest /home/wavesha/catkin_ws/build2/darknet_ros/gtest/CMakeFiles/gtest.dir/DependInfo.cmake --color=
make[3]: Leaving directory `/home/wavesha/catkin_ws/build2/darknet_ros'
make[3]: Leaving directory `/home/wavesha/catkin_ws/build2/darknet_ros'
/usr/bin/make -f CMakeFiles/clean_test_results_darknet_ros.dir/build.make CMakeFiles/clean_test_results_darknet_ros.dir/build
/usr/bin/make -f gtest/CMakeFiles/gtest.dir/build.make gtest/CMakeFiles/gtest.dir/build
make[3]: Entering directory `/home/wavesha/catkin_ws/build2/darknet_ros'
/usr/bin/python /opt/ros/indigo/share/catkin/cmake/test/remove_test_results.py /home/wavesha/catkin_ws/build2/darknet_ros/test_results/darknet_ros
make[3]: Entering directory `/home/wavesha/catkin_ws/build2/darknet_ros'
make[3]: Nothing to be done for `gtest/CMakeFiles/gtest.dir/build'.
[... 94 lines cut ...]
make[3]: Leaving directory `/home/wavesha/catkin_ws/build2/darknet_ros'
/usr/bin/cmake -E cmake_progress_report /home/wavesha/catkin_ws/build2/darknet_ros/CMakeFiles 
[100%] Built target run_tests
make[2]: Leaving directory `/home/wavesha/catkin_ws/build2/darknet_ros'
/usr/bin/cmake -E cmake_progress_start /home/wavesha/catkin_ws/build2/darknet_ros/CMakeFiles 0
make[1]: Leaving directory `/home/wavesha/catkin_ws/build2/darknet_ros'
cd /home/wavesha/catkin_ws/./build2/darknet_ros; catkin build --get-env darknet_ros | catkin env -si  /usr/bin/make run_tests --jobserver-fds=6,7 -j; cd -
Finished   << darknet_ros:make                                                 
Starting   >> darknet_ros:symlink                                              
Output     << darknet_ros:symlink /home/wavesha/catkin_ws/logs/darknet_ros/build.symlink.004.log
Linked: (/home/wavesha/catkin_ws/./devel2/.private/darknet_ros/share/darknet_ros/cmake/darknet_rosConfig.cmake, /home/wavesha/catkin_ws/./devel2/share/darknet_ros/cmake/darknet_rosConfig.cmake)
Linked: (/home/wavesha/catkin_ws/./devel2/.private/darknet_ros/share/darknet_ros/cmake/darknet_rosConfig-version.cmake, /home/wavesha/catkin_ws/./devel2/share/darknet_ros/cmake/darknet_rosConfig-version.cmake)
Linked: (/home/wavesha/catkin_ws/./devel2/.private/darknet_ros/lib/darknet_ros/darknet_ros_object_detection-test, /home/wavesha/catkin_ws/./devel2/lib/darknet_ros/darknet_ros_object_detection-test)
Linked: (/home/wavesha/catkin_ws/./devel2/.private/darknet_ros/lib/pkgconfig/darknet_ros.pc, /home/wavesha/catkin_ws/./devel2/lib/pkgconfig/darknet_ros.pc)
Finished   << darknet_ros:symlink                                              
Finished  <<< darknet_ros                [ 0.8 seconds ]                       
[build] Summary: All 1 packages succeeded!                                     
[build]   Ignored:   1 packages were skipped or are blacklisted.               
[build]   Warnings:  None.                                                     
[build]   Abandoned: None.                                                     
[build]   Failed:    None.                                                     
[build] Runtime: 0.8 seconds total.                                            
wavesha@wavesha-OptiPlex-9020:~/catkin_ws$ 

```

alexampa, 2018-05-08:
If you built the package using catkin_make you must run the tests using catkin_make run_tests_darknet_ros command.

Tkzccsk, 2018-10-20:
> If you built the package using catkin_make you must run the tests like this:
> catkin_make run_tests_darknet_ros

thank you

closed 2021-05-19 by tomlankhorst

Rao-Kai, 2024-03-25:
cy