leggedrobotics/darknet_ros #34: catkin_make error (about opencv)

wang19950602, 2017-09-20:
Hello, I got a little problem with catkin_make.
I use ubuntu 14.04 , ROS:indigo , gcc version 4.8.4
When running 'catkin_make -DCMAKE_BUILD_TYPE=Release',I encounter a problem:
[image]
The result shows that undefined reference to several opencv functions.
May I ask you how to fix this problem?
Thank you

floft, 2017-09-26:
In *src/darknet_ros/darknet_ros/CMakeLists.txt*:
find the *target_link_libraries* and add ${OpenCV_LIBS} to the list ([src](https://answers.ros.org/question/57187/opencv-undefined-reference-problem/))

floft, 2017-09-27:
Okay, so I'm on indigo and this appears to work:

    cd src/
    git clone https://github.com/superjax/cv3_bridge

In *src/darknet_ros/darknet_ros/CMakeLists.txt*:
* find the *cv_bridge* and replace with *cv3_bridge*
* find the *find_package(OpenCV REQUIRED)* and change to *find_package(OpenCV 3 REQUIRED)*
* find the *target_link_libraries(${PROJECT_NAME} ...* and add ${OpenCV_LIBRARIES} to the end of the list

Achllle, 2017-11-19:
The fix by @floft worked on Kinetic, you have to additionally go into package.xml in darknet_ros and change `cv_bridge` to `cv3_bridge`.

joyyang1215, 2018-01-19:
Add ${OpenCV_LIBS} and it works.
Thanks!

araster1010, 2018-01-21:
I have same problem as you.
I tried cv3_bridge, but it doesn't work.

In the CMakelists.txt, there are 4 target_link_libraries.
Should I add ${OpenCV_LIBS} all of them?

Could you tell me more detail about this?

mbjelonic, 2018-01-21:
You can have a look in the CMakeLists.txt of the branch devel/threads. I fixed it there. I will also merge this branch soon because the detection is much faster on this branch.

mbjelonic, 2018-01-29:
Merged the branch devel/threads.

closed 2018-01-29 by mbjelonic