leggedrobotics/darknet_ros #37: Can I use other yaml?

roskos, 2017-10-13:
Hi.
I hope you guys doing well.

I am using this ROS package on Jetson TX1 with Jetpack 2.3.1.
And the version is ubuntu 16.04 and ROS kinetic.

I am able to work this package without big troubles.
But when I switch a yaml file from yolot_voc.yaml to others such as yolo9000.yaml, I get this error.

Loading weights from /home/apsync/catkin_ws/src/darknet_ros/darknet_ros/yolo_network_config/weights/yolo.weights...Done!
terminate called after throwing an instance of 'image_transport::TransportLoadException'
  what():  Unable to load plugin for transport 'raw', error string:
Could not find library corresponding to plugin image_transport/raw_sub. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[darknet_ros-1] process has died [pid 18209, exit code -6, cmd /home/apsync/catkin_ws/devel/lib/darknet_ros/darknet_ros __name:=darknet_ros `__log:=/home/apsync/.ros/log/0d394292-b005-11e7-a848-00044b7976f5/darknet_ros-1.log].`
log file: /home/apsync/.ros/log/0d394292-b005-11e7-a848-00044b7976f5/darknet_ros-1*.log
``

This seems to be caused around image_transport.
Does anyone know how to fix it?
I retreived some ways for it but I could not find any solution.

I would appreciate if someone gives help for this problem.
Thanks.

DanLiu0623, 2018-01-25:
@roskos I met the same issue with you, did you solve it? I has two diff yaml files ,and one detect one class can run rightly, another detect two classes has same error with you,I check my rospack list, all the package is installed.I have  no idea about this problem~

tomlankhorst, 2021-05-19:
Closing due to inactivity. Feel free to reopen if you would like to add anything.

closed 2021-05-19 by tomlankhorst