leggedrobotics/darknet_ros #38: Does anyone test performance of tiny-yolo/yolo on JetsonTX2?

joyyang1215, 2017-10-19:
Could you please share your testing FPS for me?
I'm wandering the highest performance that it might achieve.
Thank you all.

kshokn2, 2017-10-20:
hi, Joy
i'm also trying on TX2 and my model is same tiny-yolo.
i had 5~6 fps, but in original darknet i had 8.5~10 fps. so i plan to apply other darknet made by Alexy.
if i test using Alexy darknet, i can have 20 fps in the same cfg file.
but after compile, that can't detect anything and fps is also 6.. now i'm trying to fix this problem..
thanks.

joyyang1215, 2017-10-20:
Cool. Thank you, I've experience almost the same fps as you.
But, I haven't tried Alexy darknet before.

OhMeuMenino, 2017-11-10:
Hi
Without ROS I had 11 FPS for Tiny-YOLO and 5 FPS for YOLO.
With ROS (darknet_ros) I just had 1 FPS.

I'm using the original Darknet

mbjelonic, 2017-11-20:
Hi,

can you please checkout the new branch called "devel/threads" and try to run it again? I heavily rearranged the code and this should also improve the speed of the detection algorithm by adding new threads. Please note that the branch is still under development and I did not add all ROS publishers in this branch. I just wanted to see if this setup improves the speed of the detection. You can check the update rate in the terminal.

Please let me know if the new code design improves also the speed in your case. Thanks in advance!

Best,
Marko

gangyahaidao, 2018-03-22:
after test AlexyAB darknet, i can get 12fps, it's faster, but it cannot work with leggedrobotics/darknet_ros,
now i am trying to fix them, mostly about them as follows:
```
CMake Error at darknet_ros_lib_generated_crop_layer_kernels.cu.o.cmake:207 (message):
  Error generating
  /home/rosrobot/catkin_ws/build/darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/./darknet_ros_lib_generated_crop_layer_kernels.cu.o
```

nesvera, 2018-06-27:
Hi, 

I am testing this package on Jetson tx2. Achieved ~= 23 fps with tiny-yolo using the onboard camera on resolution of 1280x720.

tomlankhorst, 2021-05-19:
Make sure to build in release mode. Closing for now, feel free to reopen.

closed 2021-05-19 by tomlankhorst

zerosnsa, 2024-04-30:
> 你好
> 
> 我正在 Jetson tx2 上测试此软件包。使用 tiny-yolo 以 640x360 的分辨率使用 tiny-yolo 实现 ~= 360 fps。

How did you achieve this？