leggedrobotics/darknet_ros #39: Compilation error nvcc fatal

tiagojdias, 2017-10-20:
Can anyone help me with this compilation error?

nvcc fatal   : Stray '"' character
CMake Error at darknet_ros_lib_generated_deconvolutional_kernels.cu.o.Release.cmake:222 (message):
  Error generating

Thank you.

astha736, 2017-10-27:
Hi, 

Maybe you can try removing "**set(CMAKE_CXX_FLAGS "-std=c++11 ${CMAKE_CXX_FLAGS}")**" from the darknet_ros_msgs/CMakeList.txt and add "**set( CUDA_PROPAGATE_HOST_FLAGS OFF )**" in darknet_ros_msgs/CMakeLists.txt and darknet_ros/CMakeLists.txt. 

I solved some of my issues with these modifications, including the errors for *cu.o files. I hope this helps.

gangyahaidao, 2018-03-22:
it doesnot help to solve my problem

zhanghming, 2019-04-21:
I have the same problem, and cannot solve it.
```
nvcc fatal   : Stray '"' character CMake Error at darknet_ros_lib_generated_maxpool_layer_kernels.cu.o.cmake:222 (message):
```
I don't know why

Jichen66, 2019-04-22:
I also have the same problem and cannot solve it. Can anybody do me a favor?

jianpursuit, 2019-06-18:
I also have the same problem and cannot solve it. Can anybody do me a favor?

q8wwe, 2020-03-30:
i have same problem , How to solve?

cmiller1990, 2020-10-26:
I think this is related to [this issue here](https://github.com/NVIDIA/gvdb-voxels/issues/2).

As suggested, upgrading cmake to 3.11.4 resolved the issue for me.

tomlankhorst, 2021-05-19:
Should be resolved in recent versions.

closed 2021-05-19 by tomlankhorst