leggedrobotics/darknet_ros #40: detection in low fps, screen goes dark

YuehChuan, 2017-10-31:
Hi, the package is arranged in neat and elegant way. Thanks!
But, the frame rate is too low and sometimes screen goes  dark.
I'm using PS3 eye  webcam and i5 lenovo x220 , is it normal?
also change ros.yaml [here](https://github.com/leggedrobotics/darknet_ros/blob/master/darknet_ros/config/ros.yaml#L4) to my /usb_cam/image_raw topic. (ROS usb_cam)
Did i mess up something?
thank you.
[image]

OhMeuMenino, 2017-11-10:
The black screen probably is because you lose the image.
I also have low FPS (about 1 FPS) while without darknet_ros I had 11 FPS (Tiny) and 5 (YOLO)

mbjelonic, 2017-11-20:
Hi,

can you please checkout the new branch called "devel/threads" and try to run it again? I heavily rearranged the code and this should also improve the speed of the detection algorithm by adding new threads. Please note that the branch is still under development and I did not add all ROS publishers in this branch. I just wanted to see if this setup improves the speed of the detection. You can check the update rate in the terminal.

Please let me know if the new code design improves also the speed in your case. Thanks in advance!

Best,
Marko

ddle, 2017-12-05:
regarding devel/threads branch:
> QXcbConnection: Could not connect to display
I got this error when launching via ssh (didn't happen with the master branch)

pnambiar, 2017-12-22:
Did anyone try using the devel branch  on TX2. What is the frame rate you are getting for yolo and tiny_yolo model?

DanLiu0623, 2018-01-16:
@pnambiar  Yes , I did and I got 20fps using the devel branch on TX2,but it can not show detect image, And I have not find the problem, @mbjelonic for me information that he will fix this problem in 2weeks.