leggedrobotics/darknet_ros #43: Might be that data format is being processed wrong (BGR instead of RGB)

dhiegomaga, 2017-11-19:
It took me _some hours_ to figure out why my trained network for custom object detection were not showing any bounding boxes when using in this particular implementation of darknet bridge... 

Well, it turns out that most neural networks are trained using the RGB color format (including mine), and in [this line](https://github.com/leggedrobotics/darknet_ros/blob/master/darknet_ros/src/YoloObjectDetector.cpp#L307) the image is being converted to BGR format. Changing it to: 

cam_image = cv_bridge::toCvCopy(msg, sensor_msgs::image_encodings::RGB8);

may give more accurate results - while making you look like an avatar if you're using OpenCV display, because opencv uses BGR formatting, because the alphabet..)

Most of this big networks out there are so well trained that they are robust to color channel swaps anyway, but if you're training against your own objects it might be an issue.

mbjelonic, 2017-11-20:
Thank you for the help! I changed all the the BGR8 formatting into RGB8.

closed 2017-11-20 by mbjelonic

RuijieZhu94, 2022-03-24:
> st the weig

Perhaps you can directly change the format of msg encoding after cam_image was defined.
`cv_bridge::CvImagePtr cam_image;
try ..........`
cam_image = cv_bridge::toCvCopy(msg, "rgb8");
>