leggedrobotics/darknet_ros #45: fatal error: darknet_ros_msgs/BoundingBoxes.h: No such file or directory

PLester, 2017-11-29:
1.  The error like this below:
In file included from /home/pl/catkin_ws/src/darknet_ros/darknet_ros/src/object_detector_demo.cpp:15:0:
/home/pl/catkin_ws/src/darknet_ros/darknet_ros/include/darknet_ros/YoloObjectDetector.h:34:44: fatal error: darknet_ros_msgs/BoundingBoxes.h: No such file or directory
compilation terminated.
darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/build.make:1776: recipe for target 'darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/src/object_detector_demo.cpp.o' failed
make[2]: *** [darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/src/object_detector_demo.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /home/pl/catkin_ws/src/darknet_ros/darknet_ros/src/YoloObjectDetector.cpp:10:0:
/home/pl/catkin_ws/src/darknet_ros/darknet_ros/include/darknet_ros/YoloObjectDetector.h:34:44: fatal error: darknet_ros_msgs/BoundingBoxes.h: No such file or directory
compilation terminated.
darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/build.make:1752: recipe for target 'darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/src/YoloObjectDetector.cpp.o' failed
make[2]: *** [darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/src/YoloObjectDetector.cpp.o] Error 1
CMakeFiles/Makefile2:2338: recipe for target 'darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/all' failed
make[1]: *** [darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed
2. If someone can help me solve this question, I will be very appreciate for it.
    Thank you !

klpchan, 2018-01-02:
I have same question on lunar, why ?

klpchan, 2018-01-03:
hum I solved the issue.
1) remove darknet_ros package to other place, ONLY catkin_make the darknet_ros_msg package
2) when 1 finished, the .h file is created at include folder, then add darknet_ros package to make it.
Solved~

PLester, 2018-01-03:
Thank you for your anwser, after I testify the method I will give a response/

发送自 Windows 10 版邮件<https://go.microsoft.com/fwlink/?LinkId=550986>应用

klpchan, 2018-01-18:
It's build order issue. darknet_ros_msgs should be built before darknet_ros_lib.
This issue can be solved by adding 
add_dependencies(${PROJECT_NAME}_lib ${catkin_EXPORTED_TARGETS}) 
into CMakeList file of darknet_ros.
ref : http://wiki.ros.org/catkin/CMakeLists.txt

PLester, 2018-01-18:
Thank you for your detailed answer!

luyaxige, 2018-07-28:
I add add_dependencies() after add_dependencies(), then this problem solved, only need catkin_make once. thanks for klpchan's answers!