leggedrobotics/darknet_ros #49: No cudnn in CMakeLists.txt ?

kshokn2, 2017-12-14:
Hi threre.
Thank you for your code. It is good.
I have one question in 'CMakeLists.txt'.
In Makefile of original darknet, I can choose using opencv, cudnn and etc.
So I check the 'CMakeLists.txt' in darknet_ros/darknet_ros, if i can choose this option like original darknet.
And I found that CUDNN is not including in here.
Is it right? If yes, can you help me compiling with CUDNN?

DanLiu0623, 2018-01-31:
@kshokn2 Do you have any idea about this issue? I add
`option(CUDNN "Enable CuDNN" OFF)
if (${CUDNN})
  add_definitions(-DCUDNN=1)
endif()`
refer to [link](https://github.com/ThundeRatz/ros_yolo2/blob/master/CMakeLists.txt),but it seems not work.

mbjelonic, 2018-01-31:
The CMakeLists.txt checks automatically if CUDA is installed. If you do not provide CUDA, the software runs on your CPU. In order to run the software on your GPU you simply install CUDA and the CMakeLists.txt enables the CUDA flag.

closed 2018-01-31 by mbjelonic

DanLiu0623, 2018-01-31:
I have install CUDA, but I also want to use CUDNN. I see makefile in darknet-master, So I think if I can use CUDNN in darknet-ros.
`ifeq ($(CUDNN), 1) 
COMMON+= -DCUDNN 
CFLAGS+= -DCUDNN
LDFLAGS+= -lcudnn
endif`

kshokn2, 2018-02-01:
@DanLiu0623 @mbjelonic Thank you for the answer. Also I installed CUDA and CUDNN. I tried to add a option in cmakelist, but I failed. Because I'm not good at making cmake file, I just copied lines related with CUDNN in other cmakelist many times. Anyway, It did not work.
And now I'm working other task. I will try next time..!

cmehrshad, 2020-12-23:
I have also successfully built darknet_ros to use CUDA. However, CUDNN is still not supported. CUDNN can potentially increase performance by 3x. It is nice to have it in CMakeLists.txt.