leggedrobotics/darknet_ros #51: Frame rate

pnambiar, 2017-12-22:
Did anyone try using the devel branch on TX2. What is the frame rate you are getting for yolo and tiny_yolo model?

mbjelonic, 2017-12-22:
I tried one on my tx2 and it was over 10 Hz. It is still a development branch and it does not include all features yet. I will finalize the branch soon. It would be nice if other people test it on their system for comparison.

pnambiar, 2017-12-22:
Thanks very much! I will implement it on our TX2. You package has been very useful for our project. Just that we need to get it running at a reasonably good frame rate. Thanks again

mbjelonic, 2017-12-22:
I happy that you are able to use it!
Do not run it over ssh. Currently I did not add the automatic recognition if X11 is running on the branch.

pnambiar, 2017-12-22:
ok. Thanks . Eventually we need to run it on multiple machines.

pnambiar, 2017-12-29:
tiny-yolo @  about 6 fps on TX2. Did you make any changes to settings to run at 10 fps?

pnambiar, 2017-12-29:
@mbjelonic Also, when will you be adding the capability to run over ssh?

mbjelonic, 2017-12-29:
I will add this early next week.

ddle, 2018-01-03:
@pnambiar did you try running ~/jetson_clocks.sh before darknet?
@mbjelonic I got 15-17fps for tiny-yolo, but only ~3fps for yolo, oddly.

ddle, 2018-01-04:
I realize the height and width in yolo.cfg are 608x608 instead of 416x416 which makes the difference. Changed to 416 gives me 5-6fps

mbjelonic, 2018-01-04:
@pnambiar I added the possibility to run devel/threads over ssh by adding a check if X11 is running or not. Anyway the branch is still under development and the publishers are not running yet. However, you can check the terminal now over ssh if it is running

pnambiar, 2018-01-04:
@mbjelonic  Thanks very much. Will test it.
@ddle I will try that

DanLiu0623, 2018-01-15:
@pnambiar I try it on TX2, I got 20fps use tiny-yolo only with darknet,but when I use darknet_ros,I only got 8fps，I receive the image from usb_cam.Do you have any idea of my issue?

mbjelonic, 2018-01-15:
@DanLiu0623 : Did you test with the devel/threads branch? I achieved higher performance there but I still need to integrate all the ros publishers before I can merge the branch.

pnambiar, 2018-01-16:
@mbjelonic  I was able to run devel/threads over ssh successfully.  But the topics /darknet_ros/bounding_boxes and  /darknet_ros/detection_images does not publish any data.

pnambiar, 2018-01-16:
@ddle What is the resolution of the image you are running it on ( for 15-17 fps ) ?

mbjelonic, 2018-01-16:
I had to restructure the code a lot in order to achieve the frame rate. I have to add the ros publishers again and also make sure that this is thread safe. On the branch devel/threads the reading and detection of the images are on two different threads now. I just need some time to cleanup everything.

pnambiar, 2018-01-16:
@mbjelonic Please let me know as soon as you merge the threads. Thanks

marvrez, 2018-01-17:
Hi @mbjelonic, I also got to test the devel/threads branch. When using the master branch I get 2-5fps on a TX2, but when using the devel/threads branch I get the same fps I usually would get only using the darknet framework, which is around 15-20fps. I was able to view the images through darknet's OpenCV viewer, but there were not any outputs from the ROS publishers. When will this be implemented?

pnambiar, 2018-01-18:
@marvrez  Were you able to get 15-20 fps with darknet_ros (tiny yolo) ? What is the resolution of your feed?

marvrez, 2018-01-18:
@pnambiar I got 15-20fps using the devel/threads branch in darknet_ros feeding in 416x416 pictures.

pnambiar, 2018-01-18:
@marvrez  Thanks .

mbjelonic, 2018-01-29:
Merged the branch devel/threads.

closed 2018-01-29 by mbjelonic

pnambiar, 2018-01-29:
Awesome! Thank you