leggedrobotics/darknet_ros #53: Undefined reference probelm

CCB307, 2018-01-13:
Hello, I got a little problem with catkin_make.
[image]
The result shows that undefined reference to YoloObjectDetector class constructor function.
May I ask you how to fix this problem?
Thank you
[image]

tschmidt64, 2018-01-19:
Hard to debug without more info. Did you run `catkin clean` on it beforehand?

ash22194, 2018-01-21:
I too had this problem. Cloning the repository with the recursive flag solved the problem. Check the readme.

CCB307, 2018-01-22:
@tschmidt64 @ash22194 Thank you for your answer. I have resolved the issue. ^-^!!

closed 2018-01-22 by CCB307