leggedrobotics/darknet_ros #59: Undefined references while compiling

ferhatm, 2018-02-04:
Hey, I'm getting this error while compiling ;

[ 26%] Built target darknet_ros_msgs_generate_messages_py
[ 38%] Built target darknet_ros_msgs_generate_messages_cpp
[ 56%] Built target darknet_ros_msgs_generate_messages_nodejs
[ 74%] Built target darknet_ros_msgs_generate_messages_lisp
[ 94%] Built target darknet_ros_msgs_generate_messages_eus
[ 96%] Built target darknet_ros_lib
[ 96%] Built target darknet_ros_msgs_generate_messages
[ 98%] Linking CXX executable /home/nvidia/ip_ws/devel/lib/darknet_ros/darknet_ros
CMakeFiles/darknet_ros.dir/src/yolo_object_detector_node.cpp.o: In function `main':
/home/nvidia/ip_ws/src/darknet_ros/darknet_ros/src/yolo_object_detector_node.cpp:15: undefined reference to `darknet_ros::YoloObjectDetector::YoloObjectDetector(ros::NodeHandle)'
/home/nvidia/ip_ws/src/darknet_ros/darknet_ros/src/yolo_object_detector_node.cpp:15: undefined reference to `darknet_ros::YoloObjectDetector::~YoloObjectDetector()'
/home/nvidia/ip_ws/src/darknet_ros/darknet_ros/src/yolo_object_detector_node.cpp:15: undefined reference to `darknet_ros::YoloObjectDetector::~YoloObjectDetector()'
collect2: error: ld returned 1 exit status
darknet_ros/darknet_ros/CMakeFiles/darknet_ros.dir/build.make:197: recipe for target '/home/nvidia/ip_ws/devel/lib/darknet_ros/darknet_ros' failed
make[2]: *** [/home/nvidia/ip_ws/devel/lib/darknet_ros/darknet_ros] Error 1
CMakeFiles/Makefile2:2375: recipe for target 'darknet_ros/darknet_ros/CMakeFiles/darknet_ros.dir/all' failed
make[1]: *** [darknet_ros/darknet_ros/CMakeFiles/darknet_ros.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

Can you help please ? 
Thanks

Juna2, 2018-05-30:
undefined reference to `darknet_ros::YoloObjectDetector::YoloObjectDetector(ros::NodeHandle)'

I have same issue
please save me

aveen-d, 2020-04-06:
> You need to update your gcc to a higher version. For me it worked after following this https://gist.github.com/zuyu/7d5682a5c75282c596449758d21db5ed
> and if you have multiple versions installed do
> catkin_make -DCUDA_TOOLKIT_ROOT_DIR=/usr/local/cuda -DCMAKE_C_COMPILER=gcc-6 -DCMAKE_CXX_COMPILER=g++-6 -DCMAKE_BUILD_TYPE=Release

hi i have the same error and also my gcc version 7.5.0. Can you please help me? I am attaching the error below 
CMakeFiles/darknet_ros.dir/src/yolo_object_detector_node.cpp.o: In function `main':
/home/nvidia/objd_ws/src/darknet_ros/darknet_ros/src/yolo_object_detector_node.cpp:15: undefined reference to `darknet_ros::YoloObjectDetector::YoloObjectDetector(ros::NodeHandle)'
/home/nvidia/objd_ws/src/darknet_ros/darknet_ros/src/yolo_object_detector_node.cpp:15: undefined reference to `darknet_ros::YoloObjectDetector::~YoloObjectDetector()'
/home/nvidia/objd_ws/src/darknet_ros/darknet_ros/src/yolo_object_detector_node.cpp:15: undefined reference to `darknet_ros::YoloObjectDetector::~YoloObjectDetector()'
collect2: error: ld returned 1 exit status
darknet_ros/darknet_ros/CMakeFiles/darknet_ros.dir/build.make:213: recipe for target '/home/nvidia/objd_ws/devel/lib/darknet_ros/darknet_ros' failed
make[2]: *** [/home/nvidia/objd_ws/devel/lib/darknet_ros/darknet_ros] Error 1
CMakeFiles/Makefile2:2060: recipe for target 'darknet_ros/darknet_ros/CMakeFiles/darknet_ros.dir/all' failed
make[1]: *** [darknet_ros/darknet_ros/CMakeFiles/darknet_ros.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed