leggedrobotics/darknet_ros #61: when i run the (roslaunch darknet_ros darknet_ros.launch) this error shown please give me advise!!!

fikar24fikar, 2018-02-06:
Loading weights from /home/hp-robo/catkin_ws/src/darknet_ros/darknet_ros/yolo_network_config/weights/yolo-voc.weights...Done!
[ INFO] [1517904602.426531622]: [YoloObjectDetector] USB image received.
OpenCV Error: The function/feature is not implemented (Unknown/unsupported array type) in type, file /home/hp-robo/opencv/modules/core/src/matrix.cpp, line 2034
terminate called after throwing an instance of 'cv::Exception'
  what():  /home/hp-robo/opencv/modules/core/src/matrix.cpp:2034: error: (-213) Unknown/unsupported array type in function type

[darknet_ros-1] process has died [pid 8529, exit code -6, cmd /home/hp-robo/catkin_ws/devel/lib/darknet_ros/darknet_ros __name:=darknet_ros __log:=/home/hp-robo/.ros/log/e2e45d24-0b12-11e8-a520-40b03449060a/darknet_ros-1.log].
log file: /home/hp-robo/.ros/log/e2e45d24-0b12-11e8-a520-40b03449060a/darknet_ros-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

DanLiu0623, 2018-05-23:
Do you have some idea about this issue? I met this question

weiwuhuhu, 2018-08-26:
hello, my fault is that "waiting for image" .Why I program can't receive the image?

xiachongkun, 2019-05-22:
i have the same problem. did u solve it?

Alex-Beh, 2020-07-08:
@fikar24fikar @DanLiu0623 @xiachongkun Did y'all manage to solve the issue?

panc8170, 2022-04-11:
Hey I have similar problem? How did you solve it? @Alex-Beh Can you post steps?