leggedrobotics/darknet_ros #63: Time published

gftabor, 2018-02-07:
I just noticed the detection system marks the bounding boxes as having been detected at     `boundingBoxesResults_.header.stamp = ros::Time::now();
`
Wouldn't it make more sense to publish the time from the ROS image coming in. That way the detection is accurate to the correct TF. I would expect on slow machines this to be a huge issue.

`msg.header.stamp`

https://github.com/leggedrobotics/darknet_ros/blob/master/darknet_ros/src/YoloObjectDetector.cpp#L604

ROS Image documentation
Header header        # Header timestamp should be acquisition time of image

HWiese1980, 2018-05-08:
Totally makes sense from my point of view. It has to be the stamp of the image acquisition. ros::Time::now() as stamp for the box is plain wrong.

smahajan07, 2018-07-19:
@gftabor It makes sense to stamp it with the time the image came in, and so looking at [BoundingBoxes.msg](https://github.com/leggedrobotics/darknet_ros/blob/master/darknet_ros_msgs/msg/BoundingBoxes.msg)
```
Header header
Header image_header
BoundingBox[] bounding_boxes
```
it does have the time stamp from the image too in the ``` image_header ``` .
I'm currently trying to write a node that subscribes the ``` /camera/image_color ``` and ``` /darknet_ros/bounding_boxes ``` and tries to synchronize them using [message_filters](http://wiki.ros.org/message_filters) - Time Synchronizer but it uses the default header (which is different for these topics) to synchronize. 
What would be the best way to synchronize these topics? Any suggestion?

gftabor, 2018-07-19:
So that change is actually pretty recent. This seems to be added in PR #60 in April . The really interesting bit is that the first commit to that development branch was pushed 2 days before my issue here.

gftabor, 2018-07-19:
I would imagine a solution could be implemented with this without too much difficulty.

http://wiki.ros.org/topic_tools/transform

Just republish bounding box message after swapping the headers.

bmegli, 2020-07-14:
>I would imagine a solution could be implemented with this without too much difficulty.
> http://wiki.ros.org/topic_tools/transform
> Just republish bounding box message after swapping the headers.

Assuming:
- bounding boxes published to `/camera_7/bounding_boxes`
- bbs with swapped headers published to `/camera_7/bounding_boxes_ts` 

```bash
rosrun topic_tools transform /camera_7/bounding_boxes /camera_7/bounding_boxes_ts darknet_ros_msgs/BoundingBoxes 'darknet_ros_msgs.msg.BoundingBoxes(header = m.image_header, image_header = m.header, bounding_boxes = m.bounding_boxes )' --import darknet_ros_msgs
```