leggedrobotics/darknet_ros #68: catkin_make network.h not found

waschbaer00, 2018-03-17:
Hi,

catkin_make -DCMAKE_BUILD_TYPE=Release
```

[ 98%] Built target darknet_ros_msgs_generate_messages
In file included from /home/ics_vr/softwares/darknetROS/src/darknet_ros/darknet_ros/src/yolo_object_detector_node.cpp:9:0:
/home/ics_vr/softwares/darknetROS/src/darknet_ros/darknet_ros/include/darknet_ros/YoloObjectDetector.hpp:48:21: fatal error: network.h: No such file or directory
compilation terminated.
darknet_ros/darknet_ros/CMakeFiles/darknet_ros.dir/build.make:62: recipe for target 'darknet_ros/darknet_ros/CMakeFiles/darknet_ros.dir/src/yolo_object_detector_node.cpp.o' failed
make[2]: *** [darknet_ros/darknet_ros/CMakeFiles/darknet_ros.dir/src/yolo_object_detector_node.cpp.o] Error 1
CMakeFiles/Makefile2:2484: recipe for target 'darknet_ros/darknet_ros/CMakeFiles/darknet_ros.dir/all' failed
make[1]: *** [darknet_ros/darknet_ros/CMakeFiles/darknet_ros.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j16 -l16" failed
```

How to solve this problem? Thank you.

VirginiaMenezes, 2018-04-07:
Hi, did you find a solution for this issue?
I am facing the same.

AlejoDiaz49, 2018-05-14:
Hi,

I had the same issue, it was because I downloaded the repositorie as a Zip. The darknet folder has the netwotk.h file but it was empty. I ended just clonning everything as the Readme says and works perfectly.

MilkWithCoffee, 2018-08-20:
You definitely need to clone this repo with the --recursive flag so that it also clones the darknet repo for you. Dont try to clone and integrate darknet repo manually, it did not work for me and i had the same error as you.

haosen9527, 2020-01-14:
I think you need downloaded ```darknet ```
``` shell
git clone 	https://github.com/pjreddie/darknet.git
```
~/darknet_ros/darknet