leggedrobotics/darknet_ros #71: Is the ROS wrapper compatible with yolo v3?

getaleks, 2018-04-03:


mbjelonic, 2018-04-03:
I will work on this in the next weeks.

abeyang00, 2018-04-04:
@mbjelonic ,
So i'm guessing its not compatiable with yolov3?? Can't i just use .cfg and .weights file from yolo v3?

mbjelonic, 2018-04-04:
I am not sure about the compatibility. I need to check the code for that. You can of course try to use the v3 network and see what happens.

getaleks, 2018-04-04:
I switched out the darknet path already and tried - it doesn‘t work unfortunately. And using the original darknet won‘t work because the [yolo] layers won‘t be recognised.

abeyang00, 2018-04-08:
im guessing because yolo v2 used darknet-19 wheras yolo v3 used darknet-59 (also yolo layer being different), it's not compatiable. :( I'm also trying to do ROS porting for yolo v3 but no good so far. Hopefully you succeed! @mbjelonic

Creekee, 2018-04-17:
@mbjelonic Hi, thanks for your work for yolov2. By the way, How is the progress of ros for yolov3?

lotfielhafi, 2018-04-25:
@mbjelonic: Thank very much for your work. Your package is really useful.
@getaleks: On the Yolo v3 compatibility, it seems that @kanezaki has already implemented the necessary changes in its fork: https://github.com/kanezaki/darknet_ros/commit/1f2d170bd781fa3221291211b5d6937cb2a0fa55. Maybe you should have a look.

getaleks, 2018-04-25:
> Thanks for the tip @lotfielhafi -  will be looking into this.

mbjelonic, 2018-04-25:
Hi all, 
I adapted to yolo v3. Thanks for the support and have fun with the newest version!
Best,
Marko

closed 2018-04-25 by mbjelonic