leggedrobotics/darknet_ros #76: Couldn't open file: /default/yolov2-tiny.cfg

kytu52324, 2018-04-14:
[image]
I run the package by rosrun darknet_ros darknet_ros，and it say Couldn't open file: /default/yolov2-tiny.cfg，the picture leftside.

I change the path to my computer real position，the picture rightside.

Anybody have ideal?Before the change of the path，I still got Couldn't open file: /default/yolov2-tiny.cfg...

jsdd25, 2018-05-12:
@kytu52324 Hi, have you fixed this problem? I also encountered the same problem. Can you give me some tips? I wish I could communicate with you. My e-mail is 773903267@qq.com. Thank you very much!

jsdd25, 2018-05-12:
I have fixed this problem.Thank you!

LifeBeyondExpectations, 2018-05-24:
How did you solve this problem???

QiuYilin, 2019-02-13:
how?

pushkalkatara, 2019-02-13:
Hi @QiuYilin, run the package using launch file. You can find the launch file here https://github.com/leggedrobotics/darknet_ros/blob/master/darknet_ros/launch/yolo_v3.launch

ChengXinghahaha, 2019-03-06:
@pushkalkatara I run the launch file, but it run into another problem
First section must be [net] or [network]: Resouce temporarily unavailable
........
....... shutting down processing monitor complete