leggedrobotics/darknet_ros #78: Last image stuck in detection loop

Cuky88, 2018-04-17:
When testing darknet_ros with recorded rosbags, we encounter the following problem:
1. Running darknet rosnode is fine, the net will be initialized and then the message "Waiting for image" will be printed.
2. As soon as I run the rosbag, darknet receives the image and does the inference and publishes the messages as it should
3. When I stop the rosbag, than darknet will always run the net on the last received image in loop.
There will be no message like "Waiting for image" since the [while-block](https://github.com/leggedrobotics/darknet_ros/blob/master/darknet_ros/src/YoloObjectDetector.cpp#L510) will be not left, unless the rosnode is closed - which is not. 

This beaviour is very bad. We use several cameras but it can happen that at some time no camera is recording, so darknet would do inference on the last image. Thus, it will alway publish the same bounding boxes until new images arrive. 

Did anyone else encounter this problem? How can you fix the node, so that if no new image arrives, darknet just prints "Waiting for image" and only waits?

Thx.

RhysMcK, 2018-07-05:
I am also encountering this problem. I am about to look into it, but it seems as if darknet is constantly grabbing the last received image.

elpidiovaldez, 2019-07-04:
Is there any progress towards solving this problem.  I have found various posts which all point towards darknet running at full speed even when the image has not been updated.  I found that the bounding boxes message is received repeatedly with the same frame_id, in the image header, and identical bounding boxes.  This is wasting a lot of GPU. Could darknet store the frame id and not run the NN until a new frame is received ?

elpidiovaldez, 2019-07-08:
I have made a fix which seems to solve the problem, but I still have not worked out how to publish it through Github.

Vishwajeetiitb, 2021-01-09:
can you share your repository?
I am facing same issue

Vishwajeetiitb, 2021-01-09:
> I have made a fix which seems to solve the problem, but I still have not worked out how to publish it through Github.
Can you share your repo?
I am facing the same issue
@elpidiovaldez please..