leggedrobotics/darknet_ros #79: Using Darknet-ros with nvidia jetson tx2

Francesca-Victoria-Vidal, 2018-04-17:
Hi, everyone. So I am successfully able to use YOLO with Jetson TX2 but would like to switch over to darknet-ros because I need more than one process to access the camera. I tried building this using catkin_make but I keep getting this error:

/home/nvidia/catkin_ws/src/darknet_ros/darknet_ros/include/darknet_ros/YoloObjectDetector.hpp:49:21: fatal error: network.h: No such file or directory
compilation terminated.

Has anyone faced this error before? Any advice would be appreciated.

mbjelonic, 2018-04-17:
Hi,

Did you clone this repo with recursive flag? See the readme for more information. If yes, try to clean the workspace with “catkin clean” and do “catkin build darknet_ros” again. 

Best,
Marko

Francesca-Victoria-Vidal, 2018-04-17:
No I couldn't clone it with the recursive flag because it kept saying permission denied. So I used the clone link on the page. The only issue was that the darknet @ ... folder was empty. So I had to download those and put inside the darknet folder manually. 

I am no longer seeing the initial errors I was seeing but now this is the error I have:

/home/nvidia/catkin_ws/src/darknet_ros/darknet_ros/src/YoloObjectDetector.cpp:312:64: error: ‘get_detection_boxes’ was not declared in this scope
     get_detection_boxes(l, 1, 1, demoThresh_, probs_, boxes_, 0);
                                                                ^
/home/nvidia/catkin_ws/src/darknet_ros/darknet_ros/src/YoloObjectDetector.cpp:315:34: error: ‘get_region_boxes’ was not declared in this scope
                      demoHier_, 1);
                                  ^
/home/nvidia/catkin_ws/src/darknet_ros/darknet_ros/src/YoloObjectDetector.cpp:320:63: error: cannot convert ‘box*’ to ‘detection*’ for argument ‘1’ to ‘void do_nms_obj(detection*, int, int, float)’
     do_nms_obj(boxes_, probs_, l.w * l.h * l.n, l.classes, nms);
                                                               ^
/home/nvidia/catkin_ws/src/darknet_ros/darknet_ros/src/YoloObjectDetector.cpp:327:28: error: invalid conversion from ‘int’ to ‘detection*’ [-fpermissive]
   draw_detections(display, demoDetections_, demoThresh_, boxes_, probs_, 0, dem
                            ^

mbjelonic, 2018-04-18:
You should clone it properly like in the readme. Otherwise it is hard to judge what is going on. Set up your ssh keys to solve the permission problem:

https://confluence.atlassian.com/bitbucket/set-up-an-ssh-key-728138079.html

deantorkelson, 2018-04-25:
Hi all,

I had a similar issue, but that seems to have been resolved by reinstalling boost. However, I'm still having issue regarding boost, so I can't say for sure if the issues are properly solved.

joyyang1215, 2018-06-05:
Maybe you can try
git clone --recursive https://github.com/leggedrobotics/darknet_ros.git
rather than
git clone --recursive git@github.com:leggedrobotics/darknet_ros.git
that I got "permission denied" as well.

MilkWithCoffee, 2018-08-20:
Had the same error. You definitely need to clone the repo with the --recursive flag!