leggedrobotics/darknet_ros #84: `catkin build` takes very long time

HWiese1980, 2018-05-08:
Hi everyone,

is it correct that `catkin build darknet_ros -j 8 -DCMAKE_BUILD_TYPE=Release` takes a pretty large amount of time compared to other ros packages? I've let it running for about 15 Minutes now on a Core i7-8700K with 8 jobs and it's still not finished.

Cheers

gftabor, 2018-05-08:
The main time sink is fully building darknet itself from source. I would expect building tensorflow would take a while as well if you ever tried.

The actual ros package itself should build quickly once the dependencies finish.

HWiese1980, 2018-05-08:
I see. It's 25 minutes and counting now... I'll let it run over night...

Thanks

closed 2018-05-08 by HWiese1980

HWiese1980, 2018-05-09:
Okay, looks like it took about 40 minutes to build. That's quite a lot...

KausthubK, 2020-05-02:
Thanks for asking the question... i was hoping i wasn't the only one to see this and be concerned.
@HWiese1980 any way you found to speed it up?