leggedrobotics/darknet_ros #88: How can I estimate distance from object to stereo camera based on Yolo?

ian-develop, 2018-05-17:
Hi. I'm using yolo v3 at ros.
your code is really helpful.

I wonder How can I estimate distance from object to stereo camera based on Yolo?
Yolo is object detection algorithm, but i want to know not only object name or #, but also object distance from my camera.
I already estimated a distance using detected box size (yolo is offered x, y coordi)
but its accuracy is very low.
So, How can i realize that? is there any idea?

aPonza, 2019-02-12:
Did you look further into it? After some searching I have found the package [find-object-2d](http://wiki.ros.org/find_object_2d) which estimates a pose of the recognized object, and I was hoping to combine this with YOLO detection of the object, but I'm open to ideas. I have also seen other ideas, e.g. look at [this answer](https://answers.ros.org/question/256248/how-to-get-object-coordinates-to-grip-objects/).

Imraj, 2019-04-29:
@aPonza Were you able to combine both packages?

aPonza, 2019-04-29:
I'm working on fixing camera scene recognition at the moment, not there yet actually. I was looking at the feasibility at that time.

Imraj, 2019-04-29:
Ok. This https://github.com/leggedrobotics/darknet_ros/issues/103 looks interesting. It modifies darknet_ros for the object <X,Y,Z> coordinates estimation.