leggedrobotics/darknet_ros #90: yolov3-voc.weights missing?

jokla, 2018-05-22:
Hi!

Where can I find the file yolov3-voc.weights?

I cannot download it from http://pjreddie.com/media/files/yolov3-voc.weights

Thanks!

Chrislzy1993, 2019-07-03:
hi, @jokla , man, did u save it????

marescpla, 2019-07-24:
Can someone help, please? I can't download the yolov3-voc.weights

soulslicer, 2019-08-16:
how was this solved? is this package usable anymore?

fwarmuth, 2020-01-13:
still missing, can someone upload it?

altineller, 2020-01-13:
@fwarmuth I was looking for the same thing.

https://pjreddie.com/darknet/yolo/ site does not mention the yolov3-voc-weights

but github repository does. maybe it is obsoluted?

mbjelonic, 2020-01-14:
Yes. The yolo-v3.weights is obsolete and therefore, I updated the readme. There is a http://pjreddie.com/media/files/yolov3-tiny.weights that can be used instead. There are more pre-trained weights from different data sets reported here: https://pjreddie.com/darknet/yolo/

closed 2020-01-14 by mbjelonic