leggedrobotics/darknet_ros #93: Seg Fault with YOLOV3 in Image.c

RhysMcK, 2018-05-29:
Hi,

I am having an issue when running yolo v3. After running yolo_v3.launch, shortly after I receive a segmentation fault. Compiling with gdb the culprit line seems to be at:
/darknet/src/image.c:604

604	                im.data[k*w*h + i*w + j] = data[i*step + j*c + k]/255.;

(gdb) print data
$1 = (unsigned char *) 0x23455e40 <error: Cannot access memory at address 0x23455e40>

Any thoughts on why this might be happening? 

Cheers!

RhysMcK, 2018-05-29:
Update:
I believe this issues occurs when resizing 4k resolution images down to the default 416x416. I was able to fix this problem by increasing the image resolution (608 x 608) used by darknet in the .cfg or alternatively keep the default 416 x 416 and reducing the input image resolution to Full HD quality (1920×1080)

pirobot, 2018-06-28:
We are experiencing the same segfault at the same line in image.c using an omnidirectional video stream (equirectangular) of size 3760x480 pixels.

```
ipl_into_image (src=<optimized out>, im=...) at home/patrick/jr2_catkin_ws/src/darknet_ros/darknet/src/image.c:604
604                     im.data[k*w*h + i*w + j] = data[i*step + j*c + k]/255.;
```
We have set width=3760 and height=480 in the cfg file and we get the same results whether we use yolov2-tiny or yolov2. We have tried going back to previous commits but we always get the same segfault.

Our environment is Ubuntu 16.04, CUDA 9.1.
Thanks!
patrick

RhysMcK, 2018-07-03:
Sorry for late response, we too are using Ubuntu 16.04 with CUDA 9.1

RhysMcK, 2018-07-12:
FYI, this workaround worked for me.
https://github.com/pjreddie/darknet/issues/918

vivkr, 2019-08-12:
If anyone is still having problems with seg faults, I've updated darknet to the latest commit on my fork [here](https://github.com/vivkr/darknet_ros.git) and the seg faults have stopped.

BhooshanDeshpande, 2022-04-27:
I am having the same issues, except my input image is 32 x 2048, which is causing the seg fault. Any resolution on this?