leggedrobotics/darknet_ros #95: GPU utilization

lukaszblesz, 2018-06-04:
Hi, I'm running tiny yolo on an GeForce 1080 Ti 8GB and I'm wondering that my cpu runs at over 100% ("top") while the gpu is only at 30% ("nvidia-smi"). Are these normal numbers?

RhysMcK, 2018-07-09:
I have also noticed a very high CPU usage

MilkWithCoffee, 2018-10-11:
> Hi, I'm running tiny yolo on an GeForce 1080 Ti 8GB and I'm wondering that my cpu runs at over 100% ("top") while the gpu is only at 30% ("nvidia-smi"). Are these normal numbers?

Have the same with a TitanX. GPU utilization is little (17% for yolov3-tiny), and my CPU not fully used either. Bottleneck should be somewhere else (?)

EggAgainstStone, 2019-04-02:
-gpu 0,1,2,3

jmsktm, 2019-05-06:
Similar observation as OP. gpu util < 30%, cpu util (by just the darknet_ros process) > 100%. Anybody knows a viable solution to this?

se7ven012, 2019-10-31:
I got the same issue. It still happen even if I change "GPU=0" to "GPU=1"  in Makefile.

se7ven012, 2019-11-01:
Same thing happen on my GeForce 1080 Ti with GPU utilization under 30% while CPU utilization goes above 99%.