leggedrobotics/darknet_ros #100: Process died with exit code -11

rpfly3, 2018-06-28:
```
[darknet_ros-2] process has died [pid 21702, exit code -11, cmd /home/pengfei/Documents/catkin_ws/devel/lib/darknet_ros/darknet_ros __name:=dark
net_ros __log:=/home/pengfei/.ros/log/4d4a463e-7ade-11e8-8c29-34f64b39d7e9/darknet_ros-2.log].
log file: /home/pengfei/.ros/log/4d4a463e-7ade-11e8-8c29-34f64b39d7e9/darknet_ros-2*.log
```
I continuously get send images to darknet. After sometime, it dies. I try to identify the reason with the log file. But I cannot find it. Does anyone have similar experience? Thanks a lot.

shilaimu, 2018-07-05:
I have similar errors，did you slove the problome？

shilaimu, 2018-07-12:
this errors may be caused by cfg and weights cant  match

rpfly3, 2018-07-13:
@shilaimu Check my solution here: https://github.com/pjreddie/darknet/issues/918#issuecomment-404356663

xiachongkun, 2019-05-22:
Hello, do you solve this problem? I met with the same problem. But I dont know where is wrong. Can anyone help me ?

ShiningDeadFishEyes, 2020-01-13:
still the same error . anything new ? using ubuntu 16.04 , ROS lunar , cuda 10.2

cheexuu, 2023-01-06:
Try the method I answered in this issue [#381 ](https://github.com/leggedrobotics/darknet_ros/issues/381#issuecomment-1373102550)

Vamsi-IITI, 2023-07-09:
In my case it got solved when I set batch and subdivisions to 1 in the config file. 
[image]