leggedrobotics/darknet_ros #105: using darknet_ros with ZED camera

happycity26, 2018-08-02:
Hi, I have successfully used ZED SDK with it's ROS wrapper, but I want to use it with YOLO and ROS so I came here but I do not know how to establish a connection between ZED camera and darknet_ros. I only could downloaded darknet_ros and weights for YOLOv3. I am very new to linux and ROS so if anyone can tell me how to use ZED camera or any camera with darknet_ros thoroughly I will be very happy and pleased, thank you.

catalin15dcm, 2018-08-02:
I used a kinect camera with yolo and all i have to do was to open a camera driver.
I work with freenect but i think that any other camera driver will work.

Try: 
  sudo apt-get install ros-distro-freenect-*
Where distro should be indigo, kinetic, etc

happycity26, 2018-08-03:
Hello again. My problem is solved by launching two files in two terminals. First, I launched ZED ros wrapper that is previously used and then I launched yolov3.launch with the subscriber is zed camera.

closed 2018-08-03 by mbjelonic

happycity26, 2018-08-23:
Hello @chijepark ,
You need to change two yaml files of darknet_ros files that are related to yolov3.launch file(I was using yolo version 3). If you are using different launch file just open it with gedit to see which yaml files it contains. For yolov3 yaml files are "ros.yaml" and "yolov3.yaml". I changed ros.yaml file by changing 

subscribers:

  camera reading:
    topic: /zed/left/image_raw_color
    queue_size: 1

I used left image as topic but you can use whatever you want, to see what zed publish run the below command after you run zed camera:
`rostopic list`
 Now, change yolov3.yaml file by adding image_view like follows:

image_view:

  enable_opencv: true
  use_darknet: true
  wait_key_delay: 1

After that launch zed camera first by
`$ roslaunch zed_wrapper zed.launch`
and then launching darknet ros by
`$ roslaunch darknet_ros yolo_v3.launch `
commands. I hope that would be enough but if you need any more problems related with my explanation or other issues just tell me, I will try to reply as soon as I can.

chijepark, 2018-08-24:
@happycity26  Thanks your help

but I still have problem 
I changed everything that you suggest .

**In ros.yaml file is**

subscribers:

  camera_reading:
    topic: /camera/left/image_raw_color
    queue_size: 1

actions:

  camera_reading:
    name: /darknet_ros/check_for_objects

publishers:

  object_detector:
    topic: /darknet_ros/found_object
    queue_size: 1
    latch: false

  bounding_boxes:
    topic: /darknet_ros/bounding_boxes
    queue_size: 1
    latch: false

  detection_image:
    topic: /darknet_ros/detection_image
    queue_size: 1
    latch: true

image_view:

  enable_opencv: true
  wait_key_delay: 1
  enable_console_output: true

**and yolov3.yaml file is**

yolo_model:

  config_file:
    name: yolov3.cfg
  weight_file:
    name: yolov3.weights
  threshold:
    value: 0.3
  detection_classes:
    names:
      - person
      - bicycle
      - car
      - motorbike
      - aeroplane
      - bus
      - train
      - truck
      - boat
      - traffic light
      - fire hydrant
      - stop sign
      - parking meter
      - bench
      - bird
      - cat
      - dog
[... 136 lines cut ...]
    data = m.dumps(params)
  File "/usr/lib/python2.7/xmlrpclib.py", line 638, in dumps
    dump(v, write)
  File "/usr/lib/python2.7/xmlrpclib.py", line 660, in __dump
    f(self, value, write)
  File "/usr/lib/python2.7/xmlrpclib.py", line 719, in dump_array
    dump(v, write)
  File "/usr/lib/python2.7/xmlrpclib.py", line 660, in __dump
    f(self, value, write)
  File "/usr/lib/python2.7/xmlrpclib.py", line 741, in dump_struct
    dump(v, write)
  File "/usr/lib/python2.7/xmlrpclib.py", line 660, in __dump
    f(self, value, write)
  File "/usr/lib/python2.7/xmlrpclib.py", line 719, in dump_array
    dump(v, write)
  File "/usr/lib/python2.7/xmlrpclib.py", line 660, in __dump
    f(self, value, write)
  File "/usr/lib/python2.7/xmlrpclib.py", line 664, in dump_nil
    raise TypeError, "cannot marshal None unless allow_none is enabled"
TypeError: cannot marshal None unless allow_none is enabled
Exception in thread ProcessMonitor-1 (most likely raised during interpreter shutdown):

so I search this problem 
I think I should change some more argument in yaml files 
Do you know how to solve this problem??

happycity26, 2018-08-24:
Hello again @chijepark .
I guess your problem is the syntax about gedit. You should be more careful about pressing space, and do not press tab while you are using gedit. For example, your ros.yaml file image_view and the below commands are in line which should not be. I will give you my ros.yaml and yolov3.yaml files. After trying those if problem still occurs please give me the rostopic list after you run your zed camera only.
Here is my ros.yaml file
`subscribers:

  camera_reading:
    topic: /zed/left/image_raw_color
    queue_size: 1

actions:

  camera_reading:
    name: /darknet_ros/check_for_objects

publishers:

  object_detector:
    topic: /darknet_ros/found_object
    queue_size: 1
    latch: false

  bounding_boxes:
    topic: /darknet_ros/bounding_boxes
    queue_size: 1
    latch: false

  detection_image:
    topic: /darknet_ros/detection_image
    queue_size: 1
    latch: true

image_view:

  enable_opencv: true
  wait_key_delay: 1
  enable_console_output: true`

,and here is my yolov3.yaml file
`image_view:

  enable_opencv: true
  use_darknet: true
  wait_key_delay: 1

yolo_model:

  config_file:
    name: yolov3.cfg
  weight_file:
    name: yolov3.weights
  threshold:
    value: 0.3
  detection_classes:
    names:
      - person
      - bicycle
      - car
      - motorbike
      - aeroplane
      - bus
      - train
      - truck
      - boat
      - traffic light
      - fire hydrant
      - stop sign
      - parking meter
      - bench
      - bird
      - cat
      - dog
      - horse
      - sheep
      - cow
      - elephant
      - bear
      - zebra
      - giraffe
      - backpack
      - umbrella
      - handbag
      - tie
      - suitcase
      - frisbee
      - skis
      - snowboard
      - sports ball
      - kite
      - baseball bat
      - baseball glove
      - skateboard
      - surfboard
      - tennis racket
      - bottle
      - wine glass
      - cup
      - fork
      - knife
      - spoon
      - bowl
      - banana
      - apple
      - sandwich
      - orange
      - broccoli
      - carrot
      - hot dog
      - pizza
      - donut
      - cake
      - chair
      - sofa
      - pottedplant
      - bed
      - diningtable
      - toilet
      - tvmonitor
      - laptop
      - mouse
      - remote
      - keyboard
      - cell phone
      - microwave
      - oven
      - toaster
      - sink
      - refrigerator
      - book
      - clock
      - vase
      - scissors
      - teddy bear
      - hair drier
      - toothbrush`

virraaj, 2019-07-05:
Hi.. even I am following the same stuff.  Thank you for your help. Were you able to get the 3d coordinates (for distance)  from ZED camera in this package as well?

happycity26, 2019-07-05:
I dont know about 3D coordinates, I never tried.

ironman-mark42, 2020-02-05:
Hello @happycity26 

I was implementing the same but I am experiencing the same error faced by @chijepark but then I replaced the yaml files with your yaml files and now I got this error. What should I do?

roslaunch darknet_ros yolo_v3.launch
... logging to /home/mk42/.ros/log/cbcda4dc-480b-11ea-914b-50e085400b4b/roslaunch-mk42-Alienware-m15-R2-4983.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://mk42-Alienware-m15-R2:46855/

SUMMARY
========

PARAMETERS
 * /darknet_ros/actions: None
 * /darknet_ros/bounding_boxes: None
 * /darknet_ros/camera_reading: None
 * /darknet_ros/config_file: None
 * /darknet_ros/config_path: /home/mk42/catkin...
 * /darknet_ros/detection_classes: None
 * /darknet_ros/detection_image: None
 * /darknet_ros/enable_console_output: True
 * /darknet_ros/enable_opencv: True
 * /darknet_ros/image_view: None
 * /darknet_ros/latch: True
 * /darknet_ros/name: yolov3.weights
 * /darknet_ros/names: ['person', 'bicyc...
 * /darknet_ros/object_detector: None
 * /darknet_ros/publishers: None
 * /darknet_ros/queue_size: 1
 * /darknet_ros/subscribers: None
 * /darknet_ros/threshold: None
 * /darknet_ros/topic: /darknet_ros/dete...
 * /darknet_ros/use_darknet: True
 * /darknet_ros/value: 0.3
 * /darknet_ros/wait_key_delay: 1
 * /darknet_ros/weight_file: None
 * /darknet_ros/weights_path: /home/mk42/catkin...
 * /darknet_ros/yolo_model: None
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    darknet_ros (darknet_ros/darknet_ros)

ROS_MASTER_URI=http://localhost:11311

**load_parameters: unable to set parameters (last param was [/darknet_ros/threshold=None]): cannot marshal None unless allow_none is enabled**
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/__init__.py", line 306, in main
    p.start()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py", line 279, in start
    self.runner.launch()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/launch.py", line 657, in launch
    self._setup()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/launch.py", line 644, in _setup
    self._load_parameters()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/launch.py", line 338, in _load_parameters
    r  = param_server_multi()
  File "/usr/lib/python2.7/xmlrpclib.py", line 1006, in __call__
    return MultiCallIterator(self.__server.system.multicall(marshalled_list))
  File "/usr/lib/python2.7/xmlrpclib.py", line 1243, in __call__
    return self.__send(self.__name, args)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1596, in __request
    allow_none=self.__allow_none)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1094, in dumps
    data = m.dumps(params)
  File "/usr/lib/python2.7/xmlrpclib.py", line 638, in dumps
    dump(v, write)
  File "/usr/lib/python2.7/xmlrpclib.py", line 660, in __dump
    f(self, value, write)
  File "/usr/lib/python2.7/xmlrpclib.py", line 719, in dump_array
    dump(v, write)
  File "/usr/lib/python2.7/xmlrpclib.py", line 660, in __dump
    f(self, value, write)
  File "/usr/lib/python2.7/xmlrpclib.py", line 741, in dump_struct
    dump(v, write)
  File "/usr/lib/python2.7/xmlrpclib.py", line 660, in __dump
    f(self, value, write)
  File "/usr/lib/python2.7/xmlrpclib.py", line 719, in dump_array
    dump(v, write)
  File "/usr/lib/python2.7/xmlrpclib.py", line 660, in __dump
    f(self, value, write)
  File "/usr/lib/python2.7/xmlrpclib.py", line 664, in dump_nil
    raise TypeError, "cannot marshal None unless allow_none is enabled"
TypeError: cannot marshal None unless allow_none is enabled

ironman-mark42, 2020-02-05:
This is my rostopic list 

rostopic list
/diagnostics
/rosout
/rosout_agg
/tf
/tf_static
/zed/joint_states
/zed/zed_node/confidence/camera_info
/zed/zed_node/confidence/confidence_image
/zed/zed_node/confidence/confidence_image/compressed
/zed/zed_node/confidence/confidence_image/compressed/parameter_descriptions
/zed/zed_node/confidence/confidence_image/compressed/parameter_updates
/zed/zed_node/confidence/confidence_image/compressedDepth
/zed/zed_node/confidence/confidence_image/compressedDepth/parameter_descriptions
/zed/zed_node/confidence/confidence_image/compressedDepth/parameter_updates
/zed/zed_node/confidence/confidence_image/theora
/zed/zed_node/confidence/confidence_image/theora/parameter_descriptions
/zed/zed_node/confidence/confidence_image/theora/parameter_updates
/zed/zed_node/confidence/confidence_map
/zed/zed_node/depth/camera_info
/zed/zed_node/depth/depth_registered
/zed/zed_node/depth/depth_registered/compressed
/zed/zed_node/depth/depth_registered/compressed/parameter_descriptions
/zed/zed_node/depth/depth_registered/compressed/parameter_updates
/zed/zed_node/depth/depth_registered/compressedDepth
/zed/zed_node/depth/depth_registered/compressedDepth/parameter_descriptions
/zed/zed_node/depth/depth_registered/compressedDepth/parameter_updates
/zed/zed_node/depth/depth_registered/theora
/zed/zed_node/depth/depth_registered/theora/parameter_descriptions
/zed/zed_node/depth/depth_registered/theora/parameter_updates
/zed/zed_node/disparity/disparity_image
/zed/zed_node/left/camera_info
/zed/zed_node/left/image_rect_color
/zed/zed_node/left/image_rect_color/compressed
/zed/zed_node/left/image_rect_color/compressed/parameter_descriptions
/zed/zed_node/left/image_rect_color/compressed/parameter_updates
/zed/zed_node/left/image_rect_color/compressedDepth
/zed/zed_node/left/image_rect_color/compressedDepth/parameter_descriptions
/zed/zed_node/left/image_rect_color/compressedDepth/parameter_updates
/zed/zed_node/left/image_rect_color/theora
/zed/zed_node/left/image_rect_color/theora/parameter_descriptions
/zed/zed_node/left/image_rect_color/theora/parameter_updates
/zed/zed_node/left_raw/camera_info
/zed/zed_node/left_raw/image_raw_color
/zed/zed_node/left_raw/image_raw_color/compressed
/zed/zed_node/left_raw/image_raw_color/compressed/parameter_descriptions
/zed/zed_node/left_raw/image_raw_color/compressed/parameter_updates
/zed/zed_node/left_raw/image_raw_color/compressedDepth
/zed/zed_node/left_raw/image_raw_color/compressedDepth/parameter_descriptions
/zed/zed_node/left_raw/image_raw_color/compressedDepth/parameter_updates
/zed/zed_node/left_raw/image_raw_color/theora
/zed/zed_node/left_raw/image_raw_color/theora/parameter_descriptions
/zed/zed_node/left_raw/image_raw_color/theora/parameter_updates
/zed/zed_node/odom
/zed/zed_node/parameter_descriptions
/zed/zed_node/parameter_updates
/zed/zed_node/path_map
/zed/zed_node/path_odom
/zed/zed_node/point_cloud/cloud_registered
/zed/zed_node/point_cloud/fused_cloud_registered
/zed/zed_node/pose
/zed/zed_node/pose_with_covariance
/zed/zed_node/rgb/camera_info
/zed/zed_node/rgb/image_rect_color
/zed/zed_node/rgb/image_rect_color/compressed
/zed/zed_node/rgb/image_rect_color/compressed/parameter_descriptions
/zed/zed_node/rgb/image_rect_color/compressed/parameter_updates
/zed/zed_node/rgb/image_rect_color/compressedDepth
[... 33 lines cut ...]
/zed/zed_node/right_raw/image_raw_color/compressedDepth/parameter_descriptions
/zed/zed_node/right_raw/image_raw_color/compressedDepth/parameter_updates
/zed/zed_node/right_raw/image_raw_color/theora
/zed/zed_node/right_raw/image_raw_color/theora/parameter_descriptions
/zed/zed_node/right_raw/image_raw_color/theora/parameter_updates
/zed/zed_node/stereo/image_rect_color
/zed/zed_node/stereo/image_rect_color/compressed
/zed/zed_node/stereo/image_rect_color/compressed/parameter_descriptions
/zed/zed_node/stereo/image_rect_color/compressed/parameter_updates
/zed/zed_node/stereo/image_rect_color/compressedDepth
/zed/zed_node/stereo/image_rect_color/compressedDepth/parameter_descriptions
/zed/zed_node/stereo/image_rect_color/compressedDepth/parameter_updates
/zed/zed_node/stereo/image_rect_color/theora
/zed/zed_node/stereo/image_rect_color/theora/parameter_descriptions
/zed/zed_node/stereo/image_rect_color/theora/parameter_updates
/zed/zed_node/stereo_raw/image_raw_color
/zed/zed_node/stereo_raw/image_raw_color/compressed
/zed/zed_node/stereo_raw/image_raw_color/compressed/parameter_descriptions
/zed/zed_node/stereo_raw/image_raw_color/compressed/parameter_updates
/zed/zed_node/stereo_raw/image_raw_color/compressedDepth
/zed/zed_node/stereo_raw/image_raw_color/compressedDepth/parameter_descriptions
/zed/zed_node/stereo_raw/image_raw_color/compressedDepth/parameter_updates
/zed/zed_node/stereo_raw/image_raw_color/theora
/zed/zed_node/stereo_raw/image_raw_color/theora/parameter_descriptions
/zed/zed_node/stereo_raw/image_raw_color/theora/parameter_updates

happycity26, 2021-05-14:
> I need help in take left and right image and send them to the darknet please help

It has been a while and I dont remember a thing. Good luck :)

tomlankhorst, 2021-05-19:
@youssefmohana Open a new issue when needed. Also, provide all the information you can to make the work of those trying to help you easier.

8umpk1n, 2021-12-22:
Hello, when I use zed camera to detect objects, there is a problem that red and blue are converted to each other. How can I solve it？