leggedrobotics/darknet_ros #106: Run YOLO based on actions only

Julz-s, 2018-08-03:
Hi,
I have one question.
After launching YOLO, it runs pretty much as fast as possible (similar to streaming).
However, I need to run the detector based on actions only to free the GPU for some other tasks.

Is there a simple way to achieve this or does it require major changes in the code?

Best,
Julz_s

gftabor, 2018-08-03:
I would imagine the easiest way would be having a node that reads in camera topic and passes along the image message only when necessary. A tool like this would work if you wanted to limit to 5hz etc. 
http://wiki.ros.org/topic_tools

If you want to relay images only when a boolean message is published or something it wouldn't be that different.

Julz-s, 2018-08-06:
Hi,
I'm trying to implement the latter version.

Regarding your first suggestion: I currently use the ROS image_publisher with setting the publish_rate in the launch file to 1 - but YOLO is still running at full speed.

However, darknet_ros has an action interface and I would like it to act as an action server - I send a goal containing an image and wait for the incoming result. And this works fine.
But my problem is that darknet_ros also responds to the subscribed topic, thus, running again at full speed. I also tried not to subscribe to the image topic. Then YOLO is waiting for an image (as expected) but as soon as I send an action goal, it does not only produce my result but YOLO starts running again (but I want it to do nothing until the next action request).

Best,
Julz-s

mzwang828, 2018-11-01:
Hi Julz-s,

I want to do the same thing and I'm having exactly same problem. I tried not to subscribe to a topic but once I send an action goal YOLO will keep running afterwards. Did you solve the problem? Any suggestions?

Thank you!

vivkr, 2019-08-12:
I've made changes to fix this problem [here](https://github.com/vivkr/darknet_ros). Let me know if the fix works for the both of you.