leggedrobotics/darknet_ros #107: Gray scale without any result

MhmdDimassi, 2018-08-14:
Hi i am using my laptop CORE i7 VirtualBox Ubunto 16.04 LTS , and  NVIDIA GeForce GT 740M
i have installed all files in daknet_ros according to your instructions,
and i run the test code , it runs Successfully , and this is the output :
1)
[image]
2)
[image]
3)
[image]
  
I run yolo_v3  , using the instruction : 
roslaunch darknet_ros yolo_v3.launch 
then i publish the image to the specific Topic : "/camera/rgb/image_raw" 
[image]

This is the output: 

1)
[image]
2)
[image]
 
**the output did not include the image after classification and detection and a gray screen appear**
Any idea to fix this Problem? , Thanks.

weiwuhuhu, 2018-08-26:
hello, I roslaunch darknet_ros darknet_ros.launch.The same problem is that "waiting for image".How can I resolve this problem.

MhmdDimassi, 2018-08-29:
This is a different problem.
This is because the yolo-code waiting an image to subscribe it on a specific topic  , so you should create a node that publish an image to the same topic.