leggedrobotics/darknet_ros #112: one thing for result

weiwuhuhu, 2018-08-28:
The yolo can check many things at the same time, how can i get the one thing from the results

catalin15dcm, 2018-08-28:
In the .yaml file you could delete all the names that you don't want and leave the single object that you want as a result.

You could also train yolo for a custom object from here [https://github.com/AlexeyAB/darknet#how-to-train-to-detect-your-custom-objects](url) and then add it to darknet_ros with these instructions [https://github.com/WalkingMachine/sara_wiki/wiki/Create-a-custom-dataset-(YOLO)](url)

weiwuhuhu, 2018-09-06:
Thank you very much for your help.I delete the object which I don't want in the .yaml.It's successd,but today I roslaunch again, but it dossn't work,it can't dectect the things in the .yaml. Why ?

catalin15dcm, 2018-09-17:
What is the error message?