leggedrobotics/darknet_ros #114: AlexeyAB/darknet version of darknet ROS Integration

pushkalkatara, 2018-08-31:
I find this version of darknet much optimized and fast to run yolo v3 - https://github.com/AlexeyAB/darknet/
I was trying to integrate this version with ROS based on darknet_ros, but i am facing compilation issues. Help is really appreciated. Here's the link to my fork - https://github.com/pushkalkatara/darknet_ros

MilkWithCoffee, 2018-10-02:
Nice idea, +1 for that. I am trying the same currently and will let you know if progress is made.

pushkalkatara, 2018-10-03:
[yolo_v2_class.cpp](https://github.com/AlexeyAB/darknet/blob/master/src/yolo_v2_class.cpp) - This file might help with the fetch, detect, publish thread setups.
[yolo_console_dll.cpp](https://github.com/AlexeyAB/darknet/blob/master/src/yolo_console_dll.cpp)

MilkWithCoffee, 2018-10-11:
Hi everyone,

please find my fork of this repo using **AlexeyAB's version of darknet** instead of the one by pjreddie here:
https://github.com/MilkWithCoffee/darknet_ros

pushkalkatara, 2019-01-30:
Hello,
Here's another update with alexeyAB's darknet implementation with ROS. It solves the issue of memory leak and uses AlexeyAB updates.

closed 2019-01-30 by pushkalkatara