leggedrobotics/darknet_ros #115: failed to launch darknet_ros  yolo_v3.launch

hoppss, 2018-09-03:
MY ros version is indigo.
I successfully install this realse pkg. I have modify the ros.yaml  about image topic name

when i try to launch yolo_v3.launch. the process died after loading weights file.

Loading weights from /home/robot/yolo/src/darknet_ros/darknet_ros/yolo_network_config/weights/yolov3.weights...Done!
Waiting for image.
[darknet_ros-1] process has died [pid 13522, exit code -11, cmd /home/robot/yolo/devel/lib/darknet_ros/darknet_ros __name:=darknet_ros __log:=/home/robot/.ros/log/a25df0de-af28-11e8-80b5-701ce7c05837/darknet_ros-1.log].
log file: /home/robot/.ros/log/a25df0de-af28-11e8-80b5-701ce7c05837/darknet_ros-1*.log

qboticslabs, 2019-05-22:
I am getting the same issue. I found that

Line:  detection *dets = get_network_boxes(net, buff_[0].w, buff_[0].h, demoThresh_, demoHier_, 0, 1, nboxes);

in darknet_ros/src/YoloObjectDetector.cpp. This function is failing when we try yolo3, but if we try yolo 2, this is not making an issue. Any help will be appreciated.

XueMingXiao, 2020-05-10:
i met this error too. And i found that if i changed  to launch darknet_ros.launch instead of yolo_v3.launch.  It works! I don't know why. You can have a try.

Alex-Beh, 2020-07-15:
Need to change the `network_param_file` to `yolov3.yaml` in order to use `yolo_v3.launch`.

cxhmikegithub, 2020-08-06:
> MY ros version is indigo.
> I successfully install this realse pkg. I have modify the ros.yaml about image topic name
>
> when i try to launch yolo_v3.launch. the process died after loading weights [...]

I meet the same error, have you solved it ?

ChunJyeBehBeh, 2020-08-06:
@cxhmikegithub You can try what I have said in the comment above. Change the param file.

cheexuu, 2023-01-06:
You can try the method I mentioned in [#381 ](https://github.com/leggedrobotics/darknet_ros/issues/381#issuecomment-1373102550)