leggedrobotics/darknet_ros #116: About FPS control

rcabg, 2018-09-04:
Hello,

I'm trying to control how fast YOLO is running in my TX2. I created a node that resizes and publish to YOLO images with the size and fps desired (20fps). The point is that YOLO seems to be running always as fast as it can (~25fps), so I get extra ROS messages and also it's using gpu that I need to be free for some other processing.

Is it any way to limiting this behavior?

rcabg, 2018-10-08:
hi @MilkWithCoffee, 
I'm using Tiny-Yolo-v3 with 416px.
cheers!

vivkr, 2019-08-12:
Hi @rcabg,
I faced a similar problem and I've made changes to fix it [here](https://github.com/vivkr/darknet_ros). Let me know if the fix works for you.

antondim, 2020-06-23:
Hi @vivkr darknet folder seems to be missing from your repo. How can I try your version?

vivkr, 2020-06-23:
I’ve added darknet as a sub module in my repo. You can initialise it with `git submodule update --init --recursive`. Please let me know if you have any other issues -- I’ll update the readme and repo with fixes (if any) soon.

antondim, 2020-06-23:
Thank you for fast response @vivkr !

I initially cloned it recursively and got  this:

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
remote: Repository not found.
fatal: repository 'https://github.com/BDHU/darknet.git/' not found
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

 problem keeps coming (as it was expected)

vivkr, 2020-06-23:
I've updated the darknet submodule branch to pjreddie's darknet. Unfortunately, it looks like nothing is being merged to master on that repo, and I used someone's fork of darknet because they updated pjreddie's repo to use updated CUDA functions. 

Anyway, please let me know how it goes. AlexeyAB's darknet looks a lot more promising. I could have a look into using that darknet, if that's of interest to you?

antondim, 2020-06-23:
Thing is that I would like to try this code's logic on CPU only, so I don't mind about CUDA.

Ok, I will try this and propably make a post here :) thank you.