leggedrobotics/darknet_ros #119: catkin_make problem about boost1.68.0

ShiSanChuan, 2018-09-20:
i use 'catkin_make' to make it,but it occur error in 67%,
`In file included from /opt/ros/kinetic/include/actionlib/server/handle_tracker_deleter.h:42:0,
                 from /opt/ros/kinetic/include/actionlib/server/action_server.h:51,
                 from /opt/ros/kinetic/include/actionlib/server/simple_action_server.h:42,
                 from /home/shisanchuan/catkin_ws/src/darknet_ros/darknet_ros/include/darknet_ros/YoloObjectDetector.hpp:24,
                 from /home/shisanchuan/catkin_ws/src/darknet_ros/darknet_ros/src/YoloObjectDetector.cpp:10:
/opt/ros/kinetic/include/actionlib/destruction_guard.h: In member function ‘void actionlib::DestructionGuard::destruct()’:
/opt/ros/kinetic/include/actionlib/destruction_guard.h:62:80: error: no matching function for call to ‘boost::date_time::subsecond_duration<boost::posix_time::time_duration, 1000l>::subsecond_duration(float)’
    count_condition_.timed_wait(lock, boost::posix_time::milliseconds(1000.0f));
                                                                             ^
In file included from /usr/local/include/boost/date_time/posix_time/posix_time_config.hpp:16:0,
                 from /usr/local/include/boost/date_time/posix_time/posix_time_system.hpp:13,
                 from /usr/local/include/boost/date_time/posix_time/ptime.hpp:12,
                 from /usr/local/include/boost/date_time/posix_time/posix_time_types.hpp:12,
                 from /usr/local/include/boost/thread/thread_time.hpp:11,
                 from /usr/local/include/boost/thread/detail/platform_time.hpp:11,
                 from /usr/local/include/boost/thread/pthread/condition_variable.hpp:9,
                 from /usr/local/include/boost/thread/condition_variable.hpp:16,
                 from /usr/local/include/boost/thread/condition.hpp:13,
                 from /opt/ros/kinetic/include/actionlib/server/simple_action_server.h:40,
                 from /home/shisanchuan/catkin_ws/src/darknet_ros/darknet_ros/include/darknet_ros/YoloObjectDetector.hpp:24,
                 from /home/shisanchuan/catkin_ws/src/darknet_ros/darknet_ros/src/YoloObjectDetector.cpp:10:
/usr/local/include/boost/date_time/time_duration.hpp:285:14: note: candidate: template<class T> boost::date_time::subsecond_duration<base_duration, frac_of_second>::subsecond_duration(const T&, typename boost::enable_if<boost::is_integral<Functor>, void>::type*)
     explicit subsecond_duration(T const& ss,
              ^
/usr/local/include/boost/date_time/time_duration.hpp:285:14: note:   template argument deduction/substitution failed:
/usr/local/include/boost/date_time/time_duration.hpp: In substitution of ‘template<class T> boost::date_time::subsecond_duration<base_duration, frac_of_second>::subsecond_duration(const T&, typename boost::enable_if<boost::is_integral<Functor>, void>::type*) [with T = float]’:
/opt/ros/kinetic/include/actionlib/destruction_guard.h:62:80:   required from here
/usr/local/include/boost/date_time/time_duration.hpp:285:14: error: no type named ‘type’ in ‘struct boost::enable_if<boost::is_integral<float>, void>’
In file included from /usr/local/include/boost/date_time/posix_time/posix_time_config.hpp:16:0,
                 from /usr/local/include/boost/date_time/posix_time/posix_time_system.hpp:13,
                 from /usr/local/include/boost/date_time/posix_time/ptime.hpp:12,
                 from /usr/local/include/boost/date_time/posix_time/posix_time_types.hpp:12,
                 from /usr/local/include/boost/thread/thread_time.hpp:11,
                 from /usr/local/include/boost/thread/detail/platform_time.hpp:11,
                 from /usr/local/include/boost/thread/pthread/condition_variable.hpp:9,
                 from /usr/local/include/boost/thread/condition_variable.hpp:16,
                 from /usr/local/include/boost/thread/condition.hpp:13,
                 from /opt/ros/kinetic/include/actionlib/server/simple_action_server.h:40,
                 from /home/shisanchuan/catkin_ws/src/darknet_ros/darknet_ros/include/darknet_ros/YoloObjectDetector.hpp:24,
                 from /home/shisanchuan/catkin_ws/src/darknet_ros/darknet_ros/src/YoloObjectDetector.cpp:10:
/usr/local/include/boost/date_time/time_duration.hpp:270:30: note: candidate: boost::date_time::subsecond_duration<boost::posix_time::time_duration, 1000l>::subsecond_duration(const boost::date_time::subsecond_duration<boost::posix_time::time_duration, 1000l>&)
   class BOOST_SYMBOL_VISIBLE subsecond_duration : public base_duration
`
`/opt/ros/kinetic/include/actionlib/server/simple_action_server_imp.h:396:36: error: no matching function for call to ‘boost::date_time::subsecond_duration<boost::posix_time::time_duration, 1000l>::subsecond_duration(double)’
       execute_condition_.timed_wait(lock,
                                    ^
In file included from /usr/local/include/boost/date_time/posix_time/posix_time_config.hpp:16:0,
                 from /usr/local/include/boost/date_time/posix_time/posix_time_system.hpp:13,
                 from /usr/local/include/boost/date_time/posix_time/ptime.hpp:12,
                 from /usr/local/include/boost/date_time/posix_time/posix_time_types.hpp:12,
                 from /usr/local/include/boost/thread/thread_time.hpp:11,
                 from /usr/local/include/boost/thread/detail/platform_time.hpp:11,
                 from /usr/local/include/boost/thread/pthread/condition_variable.hpp:9,
                 from /usr/local/include/boost/thread/condition_variable.hpp:16,
                 from /usr/local/include/boost/thread/condition.hpp:13,
                 from /opt/ros/kinetic/include/actionlib/server/simple_action_server.h:40,
                 from /home/shisanchuan/catkin_ws/src/darknet_ros/darknet_ros/include/darknet_ros/YoloObjectDetector.hpp:24,
                 from /home/shisanchuan/catkin_ws/src/darknet_ros/darknet_ros/src/YoloObjectDetector.cpp:10:
/usr/local/include/boost/date_time/time_duration.hpp:285:14: note: candidate: template<class T> boost::date_time::subsecond_duration<base_duration, frac_of_second>::subsecond_duration(const T&, typename boost::enable_if<boost::is_integral<Functor>, void>::type*)
     explicit subsecond_duration(T const& ss,
              ^
/usr/local/include/boost/date_time/time_duration.hpp:285:14: note:   template argument deduction/substitution failed:
/usr/local/include/boost/date_time/time_duration.hpp: In substitution of ‘template<class T> boost::date_time::subsecond_duration<base_duration, frac_of_second>::subsecond_duration(const T&, typename boost::enable_if<boost::is_integral<Functor>, void>::type*) [with T = double]’:
/opt/ros/kinetic/include/actionlib/server/simple_action_server_imp.h:396:36:   required from ‘void actionlib::SimpleActionServer<ActionSpec>::executeLoop() [with ActionSpec = darknet_ros_msgs::CheckForObjectsAction_<std::allocator<void> >]’
/opt/ros/kinetic/include/actionlib/server/simple_action_server_imp.h:128:53:   required from ‘actionlib::SimpleActionServer<ActionSpec>::SimpleActionServer(ros::NodeHandle, std::__cxx11::string, bool) [with ActionSpec = darknet_ros_msgs::CheckForObjectsAction_<std::allocator<void> >; std::__cxx11::string = std::__cxx11::basic_string<char>]’
/home/shisanchuan/catkin_ws/src/darknet_ros/darknet_ros/src/YoloObjectDetector.cpp:174:84:   required from here
/usr/local/include/boost/date_time/time_duration.hpp:285:14: error: no type named ‘type’ in ‘struct boost::enable_if<boost::is_integral<double>, void>’
In file included from /usr/local/include/boost/date_time/posix_time/posix_time_config.hpp:16:0,
                 from /usr/local/include/boost/date_time/posix_time/posix_time_system.hpp:13,
                 from /usr/local/include/boost/date_time/posix_time/ptime.hpp:12,
                 from /usr/local/include/boost/date_time/posix_time/posix_time_types.hpp:12,
                 from /usr/local/include/boost/thread/thread_time.hpp:11,
                 from /usr/local/include/boost/thread/detail/platform_time.hpp:11,
                 from /usr/local/include/boost/thread/pthread/condition_variable.hpp:9,
                 from /usr/local/include/boost/thread/condition_variable.hpp:16,
                 from /usr/local/include/boost/thread/condition.hpp:13,
                 from /opt/ros/kinetic/include/actionlib/server/simple_action_server.h:40,
                 from /home/shisanchuan/catkin_ws/src/darknet_ros/darknet_ros/include/darknet_ros/YoloObjectDetector.hpp:24,
[... cut ...]
/usr/local/include/boost/date_time/time_duration.hpp:270:30: note:   no known conversion for argument 1 from ‘double’ to ‘const boost::date_time::subsecond_duration<boost::posix_time::time_duration, 1000l>&’
/usr/local/include/boost/date_time/time_duration.hpp:270:30: note: candidate: boost::date_time::subsecond_duration<boost::posix_time::time_duration, 1000l>::subsecond_duration(boost::date_time::subsecond_duration<boost::posix_time::time_duration, 1000l>&&)
/usr/local/include/boost/date_time/time_duration.hpp:270:30: note:   no known conversion for argument 1 from ‘double’ to ‘boost::date_time::subsecond_duration<boost::posix_time::time_duration, 1000l>&&’
`

ShiSanChuan, 2018-09-23:
it can be resolve by 1.58.0.......

LC041336, 2018-12-28:
Hi guys，I met this problem recently.And also another problem.Do U have some suggestions?Waiting for your reply.

/home/err3/catkin_ws/src/darknet_ros/darknet_ros/include/darknet_ros/YoloObjectDetector.hpp:62:72: error: conflicting declaration of C function ‘void show_image_cv(image, const char*, IplImage*)’
 extern "C" void show_image_cv(image p, const char *name, IplImage *disp);
                                                                        ^
In file included from /home/err3/catkin_ws/src/darknet_ros/darknet/src/network.h:6:0,
                 from /home/err3/catkin_ws/src/darknet_ros/darknet_ros/include/darknet_ros/YoloObjectDetector.hpp:49,
                 from /home/err3/catkin_ws/src/darknet_ros/darknet_ros/src/YoloObjectDetector.cpp:10:
/home/err3/catkin_ws/src/darknet_ros/darknet/src/image.h:20:5: note: previous declaration ‘int show_image_cv(image, const char*, int)’
 int show_image_cv(image im, const char* name, int ms);
     ^
In file included from /opt/ros/kinetic/include/actionlib/server/simple_action_server.h:256:0,
                 from /home/err3/catkin_ws/src/darknet_ros/darknet_ros/include/darknet_ros/YoloObjectDetector.hpp:24,
                 from /home/err3/catkin_ws/src/darknet_ros/darknet_ros/src/YoloObjectDetector.cpp:10:
/opt/ros/kinetic/include/actionlib/server/simple_action_server_imp.h: In instantiation of ‘void actionlib::SimpleActionServer<ActionSpec>::executeLoop() [with ActionSpec = darknet_ros_msgs::CheckForObjectsAction_<std::allocator<void> >]’:
/opt/ros/kinetic/include/actionlib/server/simple_action_server_imp.h:128:53:   required from ‘actionlib::SimpleActionServer<ActionSpec>::SimpleActionServer(ros::NodeHandle, std::__cxx11::string, bool) [with ActionSpec = darknet_ros_msgs::CheckForObjectsAction_<std::allocator<void> >; std::__cxx11::string = std::__cxx11::basic_string<char>]’
/home/err3/catkin_ws/src/darknet_ros/darknet_ros/src/YoloObjectDetector.cpp:174:84:   required from here

ShiSanChuan, 2018-12-29:
> Hi guys，I met this problem recently.And also another problem.Do U have some suggestions?Waiting for your reply.
> 
> /home/err3/catkin_ws/src/darknet_ros/darknet_ros/include/darknet_ros/YoloObjectDetector.hpp:62:72: error: conflicting declaration of C function ‘void show_image_cv(image, const char*, IplImage*)’
> extern "C" void show_image_cv(image p, const char _name, IplImage _disp); ^ In file included from /home/err3/catkin_ws/src/darknet_ros/darknet/src/network.h:6:0, from /home/err3/catkin_ws/src/darknet_ros/darknet_ros/include/darknet_ros/YoloObjectDetector.hpp:49, from /home/err3/catkin_ws/src/darknet_ros/darknet_ros/src/YoloObjectDetector.cpp:10: /home/err3/catkin_ws/src/darknet_ros/darknet/src/image.h:20:5: note: previous declaration ‘int show_image_cv(image, const char_, int)’ int show_image_cv(image im, const char_ name, int ms);
> ^
> In file included from /opt/ros/kinetic/include/actionlib/server/simple_action_server.h:256:0,
> from /home/err3/catkin_ws/src/darknet_ros/darknet_ros/include/darknet_ros/YoloObjectDetector.hpp:24,
> from /home/err3/catkin_ws/src/darknet_ros/darknet_ros/src/YoloObjectDetector.cpp:10:
> /opt/ros/kinetic/include/actionlib/server/simple_action_server_imp.h: In instantiation of ‘void actionlib::SimpleActionServer::executeLoop() [with ActionSpec = darknet_ros_msgs::CheckForObjectsAction_<std::allocator >]’:
> /opt/ros/kinetic/include/actionlib/server/simple_action_server_imp.h:128:53: required from ‘actionlib::SimpleActionServer::SimpleActionServer(ros::NodeHandle, std::__cxx11::string, bool) [with ActionSpec = darknet_ros_msgs::CheckForObjectsAction_<std::allocator >; std::__cxx11::string = std::__cxx11::basic_string]’
> /home/err3/catkin_ws/src/darknet_ros/darknet_ros/src/YoloObjectDetector.cpp:174:84: required from here

hi,i found my repositories about darknet : in ~/src/darknet_ros/darknet/src/image.h ,"show_image_cv" function is return void, darknet 's author maybe fix some problem for better performance,but this repositories update slow a step, you can get a history version from  [shisanchuan/darknet_ros](https://github.com/ShiSanChuan/darknet_ros.git),it is use to UAV chasing Balloon by YOLOv3,you can use 'darknet'  folder to  repalce this work code. I hope I can help you solve your confusion.

AbdulAlkurdi, 2019-04-15:
> hi,i found my repositories about darknet : in ~/src/darknet_ros/darknet/src/image.h ,"show_image_cv" function is return void, darknet 's author maybe fix some problem for better performance,but this [...]

Thank you so much ShiSanChuan. I was having a lot of difficulties compiling but your code compiled from the 1st time.