leggedrobotics/darknet_ros #123: Is it necessary to use GPU to run the program?

xiaoxueshengyao, 2018-10-14:
I run "roslaunch darknet_ros darknete_ros.launch ",and then the image popup window shows in my screen.
But it can not work regularly,FPS is about 0.3 or less.It means a frame maybe need more than 3 seconds to be processed. My computer's processor is NUC i7-7567U without discrete GPU,so I would appreciate it if there anyone can tell me is it necessary to run the program with GPU ?

gftabor, 2018-10-14:
It sounds like it's working just fine. Your just seeing how much better Gpus are at graphics tasks than integrated graphics on cpu.

I was too lazy to find your cpus integrated graphics so I just looked up the best integrated graphics on any Intel cpu. It should get the point across. 

http://gpu.userbenchmark.com/Compare/Nvidia-GTX-1080-Ti-vs-Intel-HD-630-Desktop-Kaby-Lake/3918vsm178724

xiaoxueshengyao, 2018-10-15:
Hi,@gftabor .
Thanks for your reply.And as you said,it worked just fine,I do use a computer with Titan GPU for the program,it worked well.I want use it in a robot system to finish something like grab.Maybe I should turn my steering wheel.