leggedrobotics/darknet_ros #124: Darknet-ros process died on Jetson TX1

pushkalkatara, 2018-10-17:
I am running Yolo v3 on Jetson TX1 but after loading the model in the memory and processing some frames from the Image topic the process dies. 
I noticed in Tegra-stats, there is a constant increase in memory usage. I added a swap file of 16 GB also, but the memory fills up the swap with time and the process dies.
Some of the possible issues i thought of:
1. Memory leak between threads.
2. As the Yolo's output is around 3-5 fps and the camera's fps is 30, the image buffer is filling up and the ros publisher's queue is not rejecting frames.
3. https://github.com/ros-drivers/video_stream_opencv/issues/15

The issue is somewhere related to ROS as the same model with the config file based on Yolo v3 was successfully running without ROS.

mbjelonic, 2018-10-18:
> Small update related to the issue - https://answers.ros.org/question/220502/image-subscriber-lag-despite-queue-1/

How did you solve this issues?

pushkalkatara, 2019-01-30:
Hi @mbjelonic, I tried to solve the issue using AlexAB darknet and different multithreading implementation. The issue is resolved for me. I have pushed the code on this link - https://github.com/pushkalkatara/darknet_ros.

closed 2019-01-30 by pushkalkatara

DeepDuke, 2019-03-13:
Except for "malloc" and "free" problems, I also met 

> [darknet_ros-1] process has died [pid 13109], exit code -11

I am quite confused about this problem. I am running this package on tx2 with a Intel Realsense Camera D415.